chipsec/chipsec #581: python3-rc uefi decode regression for Lenovo firmware

hughsie, 2019-05-28:
To reproduce, checkout `python3-rc` and do:

    wget https://fwupd.org/downloads/e6619dba0394ff9f6c7918ec7b4728659f40408c-Lenovo-ThinkStation-P520-P520c-SystemFirmware-S03KT25A.cab
    gcab -x *.cab
    chipsec_util uefi decode -n S03KT25A.ROM

This results in:

    Traceback (most recent call last):
      File "~/chipsec/chipsec_util.py", line 235, in <module>
        sys.exit( main() )
      File "~/chipsec/chipsec_util.py", line 231, in main
        return chipsecUtil.main()
      File "~/chipsec/chipsec_util.py", line 198, in main
        comm.run()
      File "~/chipsec/chipsec/utilcmd/uefi_cmd.py", line 243, in run
        decode_uefi_region(_uefi, cur_dir, filename, fwtype)
      File "~/chipsec/chipsec/hal/spi_uefi.py", line 556, in decode_uefi_region
        parse_uefi_region_from_file( _uefi, fname, fwtype, fv_pth )
      File "~/chipsec/chipsec/hal/spi_uefi.py", line 538, in parse_uefi_region_from_file
        tree = build_efi_model(_uefi, rom, fwtype)
      File "~/chipsec/chipsec/hal/spi_uefi.py", line 423, in build_efi_model
        model = build_efi_tree( _uefi, data, fwtype )
      File "~/chipsec/chipsec/hal/spi_uefi.py", line 367, in build_efi_tree
        fwbin = build_efi_file_tree ( _uefi, fv_img, fwtype)
      File "~/chipsec/chipsec/hal/spi_uefi.py", line 338, in build_efi_file_tree
        fwbin.children = build_efi_modules_tree( _uefi, fwtype, fwbin.Image, fwbin.Size, fwbin.HeaderSize, polarity )
      File "~/chipsec/chipsec/hal/spi_uefi.py", line 296, in build_efi_modules_tree
        d = decompress_section_data( _uefi, "", sec_fs_name, sec.Image[sec.DataOffset:], chipsec.defines.COMPRESSION_TYPE_LZMA, True )
      File "~/chipsec/chipsec/hal/spi_uefi.py", line 73, in decompress_section_data
        uncompressed_image = _uefi.decompress_EFI_binary( compressed_name, uncompressed_name, compression_type )
      File "~/chipsec/chipsec/hal/uefi.py", line 372, in decompress_EFI_binary
        if self.cs.helper.decompress_file( compressed_name, uncompressed_name, compression_type ):
      File "~/chipsec/chipsec/helper/oshelper.py", line 549, in decompress_file
        ret = self.helper.decompress_file( CompressedFileName, OutputFileName, CompressionType )
      File "~/chipsec/chipsec/helper/linux/linuxhelper.py", line 1143, in decompress_file
        data = subprocess.check_output(decode_str,shell=True)
      File "/usr/lib64/python2.7/subprocess.py", line 223, in check_output
        raise CalledProcessError(retcode, cmd, output=output)
    subprocess.CalledProcessError: Command \'~/chipsec/chipsec_tools/compression/bin/LzmaCompress -d -o sect01_gxeo sect01_gxeo.gz\' returned non-zero exit status 1

Using `1.3.7` results in ~60 files being decompressed successfully.

hughsie, 2019-05-28:
Ohh, I also had to cherry pick a9ce1a3ad05316874e428226a17a2418886cc3ca onto that branch locally just so that LzmaCompress would be found at all..

BrentHoltsclaw, 2019-05-29:
Pushing up a patch with the cherry pick mentioned above + fix to the error.  Need to add a try except for python3 to work.  That seems to give the same results as 1.3.7.
the decode function is identifying the guid to be LZMA, however the decompress function is failing (happens outside of our implementation as well).  I'll also add a comment into the patch to note that the section data is unable to decompress.

BrentHoltsclaw, 2019-05-29:
Which PR contains the hash.  I can't seem to locate it.

hughsie, 2019-05-30:
@BrentHoltsclaw ahh, it's c31daf5cb89233e6c115dc0e9ef81d64273e2056 -- sorry about that. It just needs to be cherry picked onto the python branch.

BrentHoltsclaw, 2019-07-03:
Appears to be fixed.  Please reopen if there are farther concerns with the issue.

closed 2019-07-03 by BrentHoltsclaw