chipsec/chipsec #586: chipsec uefi decode does not work with Dell firmware

hughsie, 2019-06-03:
The `chipsec uefi decode` functionality works on HP and Lenovo firmware, but not with blobs uploaded by Dell.

I wrote a bit about it here: https://blogs.gnome.org/hughsie/2019/06/02/breaking-apart-dell-uefi-firmware-capsuleupdate-packages/

tl;dr:

    CapsuleHeader
      PE Binary
        Zlib stream
          PFS
            FVH
              PE DXEs
              PE PEIMs

Would chipsec be interested in a PR that adds the Zlib stream decompression feature? I'm happy to use the functionality I've written in lvfs-website -- https://github.com/hughsie/lvfs-website/blob/master/plugins/chipsec/__init__.py#L20 -- but equally I'd be  happy if this code was merged into chipsec. I didn't want to start on a PR if this was out of scope. I wasn't also sure which branch to target, `master` or `python3`...

Feedback welcome, thanks.

ErikBjorge, 2019-06-03:
Thanks for the information.  I think we would like to integrate the support if we can since we know it is being used by vendors.  @BrentHoltsclaw and I will look at your PR and see the best way to integrate the support.

BrentHoltsclaw, 2019-07-03:
python3-rc would be the best place for the pull request.  Do you need any help with the addition?

npmitche, 2022-12-15:
@hughsie Your link doesn't show up anymore (looks like it was moved). We're still interested in adding this functionality. Can you either refresh the link, or if you want to simply create a PR, that would work too!

npmitche, 2024-02-10:
Closing due to stale link. If this is still an issue or similar issue is identified, please open a new issue.

closed 2024-02-10 by npmitche