chipsec/chipsec #588: IO port access using native functions do not work

dcg-red-team, 2019-06-03:
chipsec_util provides 2 ways to access the io ports. The default way is by using chipsec.ko kernel object making ioctl calls. 
The second method need not use the kernel object and relies on "/dev/port" if it is exposed by the Operating system. This method of access doesn't appear to work even though the OS exposes the io ports through /dev/port as you can see in the below screenshot
the below screenshot of the failure has some print statements to narrow down the issue.
1. the first command is using IOCTL calls and read the value of ( 0x40) at address 0xf040
2. the second command ( using -n option ) prints "@" which seems to be ascii for (0x40)
Similar failure happens for native_write_io_port as well.
[image]

frinzell, 2019-06-03:
Looks like there might be a missing ord() on that io_port var.

Aaron

[image]<https://user-images.githubusercontent.com/36314681/58826912-a0ef5800-85f6-11e9-8454-b6ebc0abc5dc.png>

BrentHoltsclaw, 2019-06-03:
problem is the helper is returning a string opposed to integer

BrentHoltsclaw, 2019-06-04:
native_write_io_port also appears to be broken

BrentHoltsclaw, 2019-07-03:
#591 was merged into master closing issue.  Please reopen if there are any concerns.

closed 2019-07-03 by BrentHoltsclaw