chipsec/chipsec #589: chipsec_util decode fails,  Command '/usr/bin/chipsec_tools/compression/bin/LzmaCompress -d -o sect01_expg sect01_expg.gz' returned non-zero exit status 127

comicsheep, 2019-06-04:
It seems like there is no '/usr/bin/chipsec_tools/' directory with brand new installed chipsec 1.3.7

`[CHIPSEC] Decoding SPI ROM image from a file './H3C_fw_2019-06-03_172151.bin'
[CHIPSEC] Found SPI Flash descriptor at offset 0x0 in the binary './H3C_fw_2019-06-03_172151.bin'
/bin/sh: /usr/bin/chipsec_tools/compression/bin/LzmaCompress: No such file or directory
Traceback (most recent call last):
  File "/usr/bin/chipsec_util", line 9, in <module>
    load_entry_point('chipsec==1.3.7', 'console_scripts', 'chipsec_util')()
  File "/usr/lib64/python2.7/site-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec_util.py", line 228, in main
    return chipsecUtil.main()
  File "/usr/lib64/python2.7/site-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec_util.py", line 195, in main
    comm.run()
  File "/usr/lib64/python2.7/site-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/utilcmd/decode_cmd.py", line 124, in run
    spi_uefi.decode_uefi_region(_uefi, pth, fname, fwtype)
  File "/usr/lib64/python2.7/site-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 556, in decode_uefi_region
    parse_uefi_region_from_file( _uefi, fname, fwtype, fv_pth )
  File "/usr/lib64/python2.7/site-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 538, in parse_uefi_region_from_file
    tree = build_efi_model(_uefi, rom, fwtype)
  File "/usr/lib64/python2.7/site-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 423, in build_efi_model
    model = build_efi_tree( _uefi, data, fwtype )
  File "/usr/lib64/python2.7/site-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 367, in build_efi_tree
    fwbin = build_efi_file_tree ( _uefi, fv_img, fwtype)
  File "/usr/lib64/python2.7/site-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 338, in build_efi_file_tree
    fwbin.children = build_efi_modules_tree( _uefi, fwtype, fwbin.Image, fwbin.Size, fwbin.HeaderSize, polarity )
  File "/usr/lib64/python2.7/site-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 296, in build_efi_modules_tree
    d = decompress_section_data( _uefi, "", sec_fs_name, sec.Image[sec.DataOffset:], chipsec.defines.COMPRESSION_TYPE_LZMA, True )
  File "/usr/lib64/python2.7/site-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 73, in decompress_section_data
    uncompressed_image = _uefi.decompress_EFI_binary( compressed_name, uncompressed_name, compression_type )
  File "/usr/lib64/python2.7/site-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/hal/uefi.py", line 371, in decompress_EFI_binary
    if self.cs.helper.decompress_file( compressed_name, uncompressed_name, compression_type ):
  File "/usr/lib64/python2.7/site-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/helper/oshelper.py", line 547, in decompress_file
    ret = self.helper.decompress_file( CompressedFileName, OutputFileName, CompressionType )
  File "/usr/lib64/python2.7/site-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec/helper/linux/helper.py", line 1133, in decompress_file
    data = subprocess.check_output(decode_str,shell=True)
  File "/usr/lib64/python2.7/subprocess.py", line 575, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '/usr/bin/chipsec_tools/compression/bin/LzmaCompress -d -o sect01_expg sect01_expg.gz' returned non-zero exit status 127
[root@TENCENT64 ~/chipsec-master]# /usr/bin/chipsec_tools/compression/bin/LzmaCompress -d -o sect01_expg sect01_expg.gz
-bash: /usr/bin/chipsec_tools/compression/bin/LzmaCompress: No such file or directory
[root@TENCENT64 ~/chipsec-master]# ls /usr/bin/chipsec_t*
ls: cannot access /usr/bin/chipsec_t*: No such file or directory
[root@TENCENT64 ~/chipsec-master]# ls /usr/bin/chipsec_*
/usr/bin/chipsec_main  /usr/bin/chipsec_util
`

BrentHoltsclaw, 2019-06-04:
To help debug the issue, what steps did you use to install.

comicsheep, 2019-06-04:
yum install python-devel
yum install gcc
yum install nasm
yum install elfutils
yum install elfutils-libelf-deve
yum install kernel-devel
python setup.py install

BTW,chipsec_main works fine

BrentHoltsclaw, 2019-06-04:
#590 should fix the issue.

comicsheep, 2019-06-04:
> 
> 
> #590 should fix the issue.

It's fixed,good job~

closed 2019-06-04 by comicsheep