chipsec/chipsec #592: Chipsec EFI Shell version 1.3.7 -i command-line unrecognized

MuraliManoharShanm, 2019-06-05:
1.3.7 release version :
UEFI Shell>> python chipsec_main.py -i command-line 
returns unrecognized arguments: command-line  . 1.3.6 works fine.

mastertip (June 4 2019):
Opening chipsec-main through UEFI Shell throws error efihelper.py  (line 53) chipsec.defines.ToolType.TIANO_COMPRESS: 'TianoCompress.efi'. AttributeError: 'module' object has no attribute 'ToolType'

ErikBjorge, 2019-06-05:
I will look at this today.  Can you run 1.3.7 from the UEFI Shell with the -v flag and capture the output?  This will help us make sure we are reproducing the issue in exactly the same way.

ErikBjorge, 2019-06-05:
I'm sorry,, I misunderstood your first report with 1.3.7.

The argument 'command-line' has not been a valid command line argument 1.3.6 or 1.3.7.  With the switch to using Python argparse the command line parsing has become more restrictive (1.3.6 may have ignored invalid command line parameters).  This is expected behavior.  If you were just trying to run chipsec_main.py with the -i flag you can leave 'command-line' off.

Please also note that chipsec_main.py results may not be valid if you use the -i flag.  This skips loading of platform specific configuration data and incorrect register values may be used in testing.  This may invalidate the results.

We will look into the issue on the tip.

ErikBjorge, 2019-06-05:
I was able to reproduce the issue on 'mastertip' and we will work on fixing it.  Thanks for the report.

MuraliManoharShanm, 2019-06-05:
Thanks for the quick response. Appreciate it!

BrentHoltsclaw, 2019-07-03:
Pull request was merged into master.  Please reopen the issue if there are any more concerns.

closed 2019-07-03 by BrentHoltsclaw