chipsec/chipsec #597: Incorrect BDF for IMCn_MAIN_SMBUS for BDX and HSX

wrffrz, 2019-06-07:
In the chipsec/cfg/*.xml files, bus, dev, and func are parsed as hex numbers even without the 0x prefix and due to that the definitions for the IMCn_MAIN_SMBUS device definitions are wrong in bdx.xml and hsx.xml.

They both contain:

> <device name="IMC0_MAIN_SMBUS" bus="1" dev="19" fun="0" vid="0x8086" />
> <device name="IMC1_MAIN_SMBUS" bus="1" dev="22" fun="0" vid="0x8086" />

But the dev= attribute is parsed as hex and point to the wrong device.

These should be 0x13 and 0x16 respectively.

frinzell, 2019-06-07:
Addressed in https://github.com/chipsec/chipsec/pull/579

wrffrz, 2019-06-07:
Cool, I'd missed that PR.  Looks good and also fixes the rest of the IMCn_SMB*_n devices by using the device= attribute instead of hardcoding bdf.  Thanks.

closed 2019-06-07 by wrffrz