chipsec/chipsec #603: UEFI parsing issue

abazhaniuk, 2019-06-21:
```
> sudo python chipsec_util.py spi dump lenovo_thinkpad_x1_carbon_6.bin

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.7
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'spi' with args ['dump', 'lenovo_thinkpad_x1_carbon_6.bin']

[CHIPSEC] dumping entire SPI flash memory to 'lenovo_thinkpad_x1_carbon_6.bin'
[CHIPSEC] it may take a few minutes (use DEBUG or VERBOSE logger options to see progress)
[CHIPSEC] BIOS region: base = 0x00700000, limit = 0x00FFFFFF
[CHIPSEC] dumping 0x01000000 bytes (to the end of BIOS region)
[spi] reading 0x1000000 bytes from SPI at FLA = 0x0 (in 262144 0x40-byte chunks + 0x0-byte remainder)
[CHIPSEC] completed SPI flash dump to 'lenovo_thinkpad_x1_carbon_6.bin'
[CHIPSEC] (spi dump) time elapsed 53.079
> sudo python chipsec_util.py decode lenovo_thinkpad_x1_carbon_6.bin

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.7
[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
[CHIPSEC] Executing command 'decode' with args ['lenovo_thinkpad_x1_carbon_6.bin']

[CHIPSEC] Decoding SPI ROM image from a file 'lenovo_thinkpad_x1_carbon_6.bin'
[CHIPSEC] Found SPI Flash descriptor at offset 0x0 in the binary 'lenovo_thinkpad_x1_carbon_6.bin'
Traceback (most recent call last):
  File "chipsec_util.py", line 232, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 228, in main
    return chipsecUtil.main()
  File "chipsec_util.py", line 195, in main
    comm.run()
  File "/home/virvodva/chipsec/latest/chipsec/chipsec/utilcmd/decode_cmd.py", line 124, in run
    spi_uefi.decode_uefi_region(_uefi, pth, fname, fwtype)
  File "/home/virvodva/chipsec/latest/chipsec/chipsec/hal/spi_uefi.py", line 569, in decode_uefi_region
    _uefi.parse_EFI_variables( nvram_fname, region_data, False, fwtype )
  File "/home/virvodva/chipsec/latest/chipsec/chipsec/hal/uefi.py", line 359, in parse_EFI_variables
    efi_vars = uefi_platform.EFI_VAR_DICT[ self._FWType ]['func_getefivariables']( efi_vars_store )
  File "/home/virvodva/chipsec/latest/chipsec/chipsec/hal/uefi_platform.py", line 832, in EFIvar_EVSA
    g = guid_str(guid0, guid1, guid2, guid3)
  File "/home/virvodva/chipsec/latest/chipsec/chipsec/hal/uefi_common.py", line 493, in guid_str
    guid = uuid.UUID(bytes=guid0[::-1] + guid1[::-1] + guid2[::-1] + guid3[:2] + guid3[-6:])
TypeError: 'int' object has no attribute '__getitem__'
> ls -lah lenovo_thinkpad_x1_carbon_6.bin
-rw-r--r-- 1 root root 16M Jun 21 13:03 lenovo_thinkpad_x1_carbon_6.bin
> 
```

BrentHoltsclaw, 2019-06-21:
Can you check to see if #604 addresses the issue?

abazhaniuk, 2019-06-28:
checked, closed this one

closed 2019-06-28 by abazhaniuk