chipsec/chipsec #606: chipsec_util with no command/non-existing command returns error

maggaw, 2019-06-25:
Hi All, 

I have run the chipsec_util with no command and with non-existing command on Windows and in both cases I got unhandled UnboundLocalError (see below).

git\chipsec>python chipsec_util.py

Traceback (most recent call last):
  File "chipsec_util.py", line 232, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 228, in main
    return chipsecUtil.main()
  File "chipsec_util.py", line 206, in main
    return comm.ExitCode
UnboundLocalError: local variable 'comm' referenced before assignment

Shouldn't the ExitCode be returned only if the command was provided?

I hope that it is a good place to ask.

Regards,
Magdalena

ErikBjorge, 2019-06-25:
Good catch.  We missed this when verifying the patch.  The ExitCode should be 0 because chipsec_util ran correctly.  We will fix this before the next release.

hughsie, 2019-06-25:
Err, 0 is success. Can't it return something like `2` for command not found? Otherwise users of chipsec running chipsec with a new command don't know if the command succeeded, or if the version was too old to support the new command...

    [hughsie@localhost chipsec (python3-rc %)]$ ./chipsec_util.py foo
    [CHIPSEC] Version 1.3.7
    ERROR: Unknown command 'foo'
    [hughsie@localhost chipsec (python3-rc %)]$ echo $?
    0

ErikBjorge, 2019-06-25:
If a command is not defined it should return a value other than 0.  For no command present I think it should return 0 and display the list of commands.

BrentHoltsclaw, 2019-06-25:
I believe the pull request fixes the issue and represents the logic @ErikBjorge is describing

BrentHoltsclaw, 2019-10-10:
Appears to be fixed.  Please reopen if necessary.

closed 2019-10-10 by BrentHoltsclaw