chipsec/chipsec #608: chipsec_util idt and gdt functions cause bad address error and kernel stack backtrace

wrffrz, 2019-06-27:
With the current version of chipsec, I'm getting kernel errors when I try to use the chipsec_util idt and gdt functions.

$ sudo python chipsec_util.py idt

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.7
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'idt' with args []

[CHIPSEC] Dumping IDT of 4 CPU threads
Traceback (most recent call last):
  File "chipsec_util.py", line 232, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 228, in main
    return chipsecUtil.main()
  File "chipsec_util.py", line 195, in main
    comm.run()
  File "/home/user/Desktop/chipsec/chipsec/utilcmd/desc_cmd.py", line 47, in run
    self.cs.msr.IDT_all( 4 )
  File "/home/user/Desktop/chipsec/chipsec/hal/msr.py", line 164, in IDT_all
    self.IDT( tid, num_entries )
  File "/home/user/Desktop/chipsec/chipsec/hal/msr.py", line 157, in IDT
    return self.dump_Descriptor_Table( cpu_thread_id, DESCRIPTOR_TABLE_CODE_IDTR, num_entries )
  File "/home/user/Desktop/chipsec/chipsec/hal/msr.py", line 136, in dump_Descriptor_Table
    dt = self.helper.read_physical_mem( pa, limit + 1 )
  File "/home/user/Desktop/chipsec/chipsec/helper/oshelper.py", line 228, in read_physical_mem
    ret = self.helper.read_phys_mem( (phys_address>>32)&0xFFFFFFFF, phys_address&0xFFFFFFFF, length )
  File "/home/user/Desktop/chipsec/chipsec/helper/linux/helper.py", line 374, in read_phys_mem
    return self.__mem_block(length)
  File "/home/user/Desktop/chipsec/chipsec/helper/linux/helper.py", line 350, in __mem_block
    return self.dev_fh.read(sz)
IOError: [Errno 14] Bad address

$ sudo python chipsec_util.py gdt

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.7
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'gdt' with args []

[CHIPSEC] Dumping GDT of 4 CPU threads
Traceback (most recent call last):
  File "chipsec_util.py", line 232, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 228, in main
    return chipsecUtil.main()
  File "chipsec_util.py", line 195, in main
    comm.run()
  File "/home/user/Desktop/chipsec/chipsec/utilcmd/desc_cmd.py", line 68, in run
    self.cs.msr.GDT_all( 4 )
  File "/home/user/Desktop/chipsec/chipsec/hal/msr.py", line 167, in GDT_all
    self.GDT( tid, num_entries )
  File "/home/user/Desktop/chipsec/chipsec/hal/msr.py", line 160, in GDT
    return self.dump_Descriptor_Table( cpu_thread_id, DESCRIPTOR_TABLE_CODE_GDTR, num_entries )
  File "/home/user/Desktop/chipsec/chipsec/hal/msr.py", line 136, in dump_Descriptor_Table
    dt = self.helper.read_physical_mem( pa, limit + 1 )
  File "/home/user/Desktop/chipsec/chipsec/helper/oshelper.py", line 228, in read_physical_mem
    ret = self.helper.read_phys_mem( (phys_address>>32)&0xFFFFFFFF, phys_address&0xFFFFFFFF, length )
  File "/home/user/Desktop/chipsec/chipsec/helper/linux/helper.py", line 374, in read_phys_mem
    return self.__mem_block(length)
[... 41 lines cut ...]
Jun 27 10:02:08 dell-e7470 kernel: [  408.834452] Call Trace:
Jun 27 10:02:08 dell-e7470 kernel: [  408.834470]  ? __handle_mm_fault+0x3fd/0x580
Jun 27 10:02:08 dell-e7470 kernel: [  408.834479]  ioremap_nocache+0x17/0x20
Jun 27 10:02:08 dell-e7470 kernel: [  408.834494]  my_xlate_dev_mem_ptr+0x49/0x80 [chipsec]
Jun 27 10:02:08 dell-e7470 kernel: [  408.834504]  read_mem+0xcb/0x240 [chipsec]
Jun 27 10:02:08 dell-e7470 kernel: [  408.834515]  ? security_file_permission+0xa0/0xb0
Jun 27 10:02:08 dell-e7470 kernel: [  408.834527]  __vfs_read+0x1b/0x40
Jun 27 10:02:08 dell-e7470 kernel: [  408.834535]  vfs_read+0x8e/0x130
Jun 27 10:02:08 dell-e7470 kernel: [  408.834541]  ksys_read+0x55/0xc0
Jun 27 10:02:08 dell-e7470 kernel: [  408.834548]  __x64_sys_read+0x1a/0x20
Jun 27 10:02:08 dell-e7470 kernel: [  408.834557]  do_syscall_64+0x5a/0x120
Jun 27 10:02:08 dell-e7470 kernel: [  408.834568]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Jun 27 10:02:08 dell-e7470 kernel: [  408.834574] RIP: 0033:0x7fd04b82f081
Jun 27 10:02:08 dell-e7470 kernel: [  408.834575] Code: fe ff ff 48 8d 3d 67 9c 0a 00 48 83 ec 08 e8 a6 4c 02 00 66 0f 1f 44 00 00 48 8d 05 81 08 2e 00 8b 00 85 c0 75 13 31 c0 0f 05 <48> 3d 00 f0 ff ff 77 57 f3 c3 0f 1f 44 00 00 41 54 55 49 89 d4 53
Jun 27 10:02:08 dell-e7470 kernel: [  408.834705] RSP: 002b:00007ffd3f6dce68 EFLAGS: 00000246 ORIG_RAX: 0000000000000000
Jun 27 10:02:08 dell-e7470 kernel: [  408.834712] RAX: ffffffffffffffda RBX: 000056554fbe3ab0 RCX: 00007fd04b82f081
Jun 27 10:02:08 dell-e7470 kernel: [  408.834716] RDX: 0000000000001000 RSI: 000056554ffb7cb0 RDI: 0000000000000003
Jun 27 10:02:08 dell-e7470 kernel: [  408.834721] RBP: 0000000000000d68 R08: 0000000000000b40 R09: 00007fd04863cf6c
Jun 27 10:02:08 dell-e7470 kernel: [  408.834726] R10: 000056554fbe3b90 R11: 0000000000000246 R12: 00007fd04bb06760
Jun 27 10:02:08 dell-e7470 kernel: [  408.834730] R13: 00007fd04bb072a0 R14: 0000000000000d68 R15: 0000000000000080
Jun 27 10:02:08 dell-e7470 kernel: [  408.834738] ---[ end trace bb50680b46ff327f ]---
Jun 27 10:02:08 dell-e7470 kernel: [  408.834744] ioremap: invalid physical address 651500001000
Jun 27 10:02:08 dell-e7470 kernel: [  408.834753] chipsec read_mem 427: xlate FAIL, p: 651500001000
Jun 27 10:02:15 dell-e7470 kernel: [  415.073128] chipsec cleanup_module 1786: Destroying chipsec device
Jun 27 10:02:15 dell-e7470 kernel: [  415.073223] chipsec cleanup_module 1788: exit

BrentHoltsclaw, 2019-06-28:
Within the driver, we are calling virt_to_phys to translate the address returned by the call.  virt_to_phys does not seem to work with [kmalloc addresses](https://stackoverflow.com/questions/5748492/is-there-any-api-for-determining-the-physical-address-from-virtual-address-in-li).

Sae86, 2022-05-09:
This issue seems to have been fixed by #1208 , closing as fixed. Please feel free to reopen if needed.

closed 2022-05-09 by Sae86