chipsec/chipsec #609: Decode issue with Apple EFI firmware

abazhaniuk, 2019-06-30:
Doesn't decode Apple EFI firmware. at DXE phase doesn't apply decompression (lzma) to decompress FV. binwalk help to debug. 
[mac_spi_image.tar.gz](https://github.com/chipsec/chipsec/files/3343185/mac_spi_image.tar.gz)

0xfede7c8, 2019-07-18:
Sorry, I have seen that you referenced a commit here.
I have tested this and it seems not to be fixed. 
Even more it seems to be working less than before:

With the image provided here and chipsec 1.3.6 I get 254 executabled and with latest chipsec on master I get 74. This including the recent commit.

BrentHoltsclaw, 2019-07-18:
Thanks for the input, I'll see if i can pinpoint the issue and provide a new patch.

BrentHoltsclaw, 2019-07-18:
@0xfede7c8 can you check with #628?  I still need to validate that it did not break other images, but it appears to be working for the spi image provided above.

frinzell, 2021-02-22:
Appears fixed and merged. Closing. Reopen if needed.

closed 2021-02-22 by frinzell