chipsec/chipsec #612: cpu info

abazhaniuk, 2019-06-30:
when running cpu info in Kabylake system got:

```
$ sudo python chipsec_util.py cpu info

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.7
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'cpu' with args ['info']

[CHIPSEC] CPU information:
          Hyper-Threading         : Enabled
          CPU cores per package   : 4
          CPU threads per core    : 2
          CPU threads per package : 8
[helper] Error in va2pa: PA higher that max physical address: VA (0xFFFFFFFEF7AB4018) -> PA (0x00006B41B7AB4018)
[mem] Looks like VA (0xFFFFFFFEF7AB4018) not mapped
ERROR: Can't find UEFI ConfigurationTable
[CHIPSEC] (cpu) time elapsed 1.932
virvodva@ubuntu-dev:~/chipsec/latest/chipsec$ sudo dmesg -c
[32967.669330] Chipsec module loaded 
[32967.669334] ** This module exposes hardware & memory access, **
[32967.669336] ** which can effect the secure operation of      **
[32967.669337] ** production systems!! Use for research only!   **
[32968.570728] ioremap error for 0x4ff2b000-0x4ff2c000, requested 0x2, got 0x0
[32968.570738] ioremap error for 0x4ff2c000-0x4ff2d000, requested 0x2, got 0x0
[32968.571565] ioremap error for 0x4ff3d000-0x4ff3e000, requested 0x2, got 0x0
[32968.572389] ioremap error for 0x4ff4e000-0x4ff4f000, requested 0x2, got 0x0
[32968.572394] ioremap error for 0x4ff4f000-0x4ff50000, requested 0x2, got 0x0
[32968.572399] ioremap error for 0x4ff50000-0x4ff51000, requested 0x2, got 0x0
```

abazhaniuk, 2019-06-30:
same for ACPI:
```
$ sudo python chipsec_util.py acpi list

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.7
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'acpi' with args ['list']

[helper] Error in va2pa: PA higher that max physical address: VA (0xFFFFFFFEF7AB4018) -> PA (0x00006B41B7AB4018)
[mem] Looks like VA (0xFFFFFFFEF7AB4018) not mapped
ERROR: Can't find UEFI ConfigurationTable
[CHIPSEC] Enumerating ACPI tables..
 - APIC: 0x000000004FFE9000
 - ASF!: 0x000000004FFAA000
 - BATB: 0x000000004FFBA000
 - BGRT: 0x000000004FFA8000
 - BOOT: 0x000000004FFBB000
 - DBG2: 0x000000004FFAE000
 - DBGP: 0x000000004FFAF000
 - DMAR: 0x000000004FFAC000
 - DSDT: 0x000000004FFC0000
 - ECDT: 0x000000004FFE7000
 - FACP: 0x000000004FFEB000
 - FPDT: 0x000000004FFA9000
 - HPET: 0x000000004FFEA000
 - LPIT: 0x000000004FFB4000
 - MCFG: 0x000000004FFE8000
 - MSDM: 0x000000004FFAD000
 - NHLT: 0x000000004FFAB000
 - RSDT: 0x000000004FFB2000
 - SSDT: 0x000000004FFF3000, 0x000000004FFF2000, 0x000000004FFED000, 0x000000004FFEC000, 0x000000004FFBE000, 0x000000004FFBC000, 0x000000004FFB7000, 0x000000004FFB6000, 0x000000004FFB5000, 0x000000004FFFD000, 0x000000004FFB1000, 0x000000004FFB0000
 - TPM2: 0x000000004FFF1000
 - UEFI: 0x000000004FF53000, 0x000000004FF2B000
 - WSMT: 0x000000004FFB3000
[CHIPSEC] (acpi) time elapsed 1.920
```

ErikBjorge, 2019-11-06:
Do you want to see if PR #713 fixes the issue for you?  In most cases the OS does not map the reserved memory range so we have to do some relative address computations.

closed 2026-10-12 by dscott90