chipsec/chipsec #618: Is the DCI address different in Coffee Lake?

AlanCASwell, 2019-07-05:
Hi,

We have tested the Linux UEFI Validation (https://01.org/zh/linux-uefi-validation/) which includes the chipsec tool in our Coffee Lake platform and got the following error. And our RD found that the DCI Port ID in Coffee Lake is different than others (in Coffee Lake, it is 0x71; and in Skylake, it is 0xB8).

So, the question would be, "can the chipsec tool support Coffee Lake platform in this feature?".

Thank you.
Best Regards,
Alan

== Message outputted from the chipsec tool ==
[-]chipsec.modules.debugenabled... 1 failurescid:image002.png@01D51567.1AD13270
[*] running module: chipsec.modules.debugenabled 
[+] imported: chipsec.modules.debugenabled 
[*] Module path: /usr/lib/python2.7/site-packages/chipsec/modules/debugenabled.pyc 
[x][ ======================================================================= 
[x][ Module: Debug features test 
[x][ ======================================================================= 
[X] Checking IA32_DEBUG_INTERFACE msr status 
[cpu] # of logical CPUs: 4 
[cpu0] RDMSR( 0xc80 ): EAX = 0x40000000, EDX = 0x00000000 
[cpu1] RDMSR( 0xc80 ): EAX = 0x40000000, EDX = 0x00000000 
[cpu2] RDMSR( 0xc80 ): EAX = 0x40000000, EDX = 0x00000000 
[cpu3] RDMSR( 0xc80 ): EAX = 0x40000000, EDX = 0x00000000 
[+] CPU IA32_DEBUG_INTERFACE is disabled 
[X] Checking DCI register status 
[pci] reading B/D/F: 0/31/1, offset: 0x00, value: 0xFFFFFFFF 
[pci] reading B/D/F: 0/31/1, offset: 0xE1, value: 0xFF 
[pci] writing B/D/F: 0/31/1, offset: 0xE1, value: 0xFE 
[pci] reading B/D/F: 0/31/1, offset: 0x10, value: 0xFFFFFFFF 
[mmio] SBREGBAR: 0x00000000FF000000 (size = 0x1000000) 
[mmio] 0xFF000000 + 0x00**_B8_**0004 = 0xFFFFFFFF 
[pci] reading B/D/F: 0/31/1, offset: 0x00, value: 0xFFFFFFFF 
[pci] reading B/D/F: 0/31/1, offset: 0xE1, value: 0xFF 
[pci] writing B/D/F: 0/31/1, offset: 0xE1, value: 0xFF 
[*] DCI Control Register = 0xFFFFFFFF 
[-] DCI Debug is enabled 
[-] FAILED: One or more of the debug checks have failed and a debug feature is enabled

== DCI Port ID definition ==
[image]

ErikBjorge, 2019-07-08:
A couple of issues will need to be fixed to get this working on CFL.  A fixed base address will need to be used for P2SB and the port ID will need to be updated as well.  Thanks for filing this issue and we will get this fixed soon.

AlanCASwell, 2019-07-09:
Thanks, ErikBjorge.

martingalvan, 2019-10-31:
Oddly, enough, I'm not seeing `ECTRL` defined in the 300 series PCH datasheet (nor in the "on-package" on).

kevinsun49, 2022-06-22:
Closing this issue as it seems to be fixed.

closed 2022-06-22 by kevinsun49