chipsec/chipsec #620: Python error when executin rogue_mmio_bar_run() and smm_ptr

dummys, 2019-07-09:
Hello,

I'm trying to use the mode `tools.smm.rogue_mmio_bar` but I get this error:

```
C:\Users\test\Downloads\chipsec-master>python chipsec_main.py -m tools.smm.rogue_mmio_bar
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.7
[CHIPSEC] Arguments: -m tools.smm.rogue_mmio_bar

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Windows 10 10.0.17134 AMD64
[CHIPSEC] Platform: Mobile 8th Generation Core Processor (Kabylake U-Quad Core)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 5914
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 9D4E

[+] loaded chipsec.modules.tools.smm.rogue_mmio_bar
[*] running loaded modules ..

[*] running module: chipsec.modules.tools.smm.rogue_mmio_bar
[x][ =======================================================================
[x][ Module: experimental tool to help checking for SMM MMIO BAR issues
[x][ =======================================================================
[*] discovering PCIe devices..
[*] testing MMIO of PCIe devices:
    00:00.0
    00:02.0
    00:04.0
    00:13.0
    00:14.0
    00:14.2
    00:15.0
    00:15.1
    00:16.0
    00:1C.0
    00:1C.4
    00:1D.0
    00:1F.0
    00:1F.2
    00:1F.3
    00:1F.4
    00:1F.6
    01:00.0
    3B:00.0
[*] allocated memory range : 0x0000000094819000 (0x20000 bytes)
[*] MMIO relocation address: 0x0000000094820000

[*] enumerating device 00:00.0 MMIO BARs..
[*] enumerating device 00:02.0 MMIO BARs..
[*] enumerating device 00:04.0 MMIO BARs..
[*] found MMIO BAR +0x10 (base 0x0000000094840000, size 0x8000)
[*] copying BAR 0x94840000 > 0x94820000
> SMI# 00: data 00, func (ECX) 0
  relocating BAR 0x94840004
  new BAR: 0x94820004
ERROR: Exception occurred during chipsec.modules.tools.smm.rogue_mmio_bar.run(): 'unpack requires a string argument of length 56'

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            1.628
[CHIPSEC] Modules total           1
[CHIPSEC] Modules failed to run   1:
ERROR: chipsec.modules.tools.smm.rogue_mmio_bar
[CHIPSEC] Modules passed          0:
[CHIPSEC] Modules information     0:
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   0:
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  0:
[CHIPSEC] Modules with exceptions 1:
ERROR: chipsec.modules.tools.smm.rogue_mmio_bar
[CHIPSEC] *****************************************************************
```

and the error when trying to run `smm_ptr.run()`:

```
C:\Users\test\Downloads\chipsec-master>python chipsec_main.py -m tools.smm.smm_ptr -a fuzz,0:0x40
[*] Use of the -a command no longer needs to have arguments concatenated with ','
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.7
[CHIPSEC] Arguments: -m tools.smm.smm_ptr -a fuzz,0:0x40

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Windows 10 10.0.17134 AMD64
[CHIPSEC] Platform: Mobile 8th Generation Core Processor (Kabylake U-Quad Core)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 5914
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 9D4E

[+] loaded chipsec.modules.tools.smm.smm_ptr
[*] running loaded modules ..

[*] running module: chipsec.modules.tools.smm.smm_ptr
[*] Module arguments (2):
['fuzz', '0:0x40']
[x][ =======================================================================
[x][ Module: A tool to test SMI handlers for pointer validation vulnerabilies
[x][ =======================================================================
Usage: chipsec_main -m tools.smm.smm_ptr [ -a <mode>,<config_file>|<smic_start:smic_end>,<size>,<address> ]
  mode          SMI handlers testing mode
    = config    use SMI configuration file <config_file>
    = fuzz      fuzz all SMI handlers with code in the range <smic_start:smic_end>
    = fuzzmore  fuzz mode + pass '2nd-order' pointers within buffer to SMI handlers
  size          size of the memory buffer (in Hex)
  address       physical address of memory buffer to pass in GP regs to SMI handlers (in Hex)
    = smram     pass address of SMRAM base (system may hang in this mode!)

[*] Allocated memory buffer (to pass to SMI handlers)       : 0x0000000095A36000
[*] Allocated 2nd buffer (address will be in the 1st buffer): 0x0000000095A34000

[*] Configuration
    SMI testing mode          : fuzz
    Range of SMI codes (B2)   : 0x00:0x40
    Memory buffer pointer     : 0x0000000095A36000 (address passed in GP regs to SMI)
    Filling/checking memory?  : YES
      Second buffer pointer   : 0x0000000095A34000 (address written to memory buffer)
      Number of bytes to fill : 0x500
      Byte to fill with       : 0x11
    Additional options (can be changed in the source code):
      Fuzzing SMI functions in ECX?          : 1
      Max value of SMI function in ECX       : 0x10
      Max value of SMI data (B3)             : 0x100
      Max offset of the pointer in the buffer: 0x20
      Passing pointer in all GP registers?   : 0
      Default values of the registers        : 0x5A5A5A5A5A5A5A5A
      Dump all register values every SMI     : 1
      Bail on first detection                : 1

[*] >>> Fuzzing SMI handlers..
[*] AX in RAX will be overwridden with values of SW SMI ports 0xB2/0xB3
    DX in RDX will be overwridden with value 0x00B2

[*] Setting values of general purpose registers to 0x5A5A5A5A5A5A5A5A

[*] reloading buffer with PTR at offset 0x0..
[*] writing 0x500 bytes at 0x0000000095A36000

[*] fuzzing SMI# 0x00 (data: 0x00)
 >> function (RCX): 0x0000000000000000
    RBX: 0x0000000095A36000
    > SMI 00 (data: 00)
      RAX: 0x5A5A5A5A5A5A5A5A
      RBX: 0x0000000095A36000
      RCX: 0x0000000000000000
      RDX: 0x5A5A5A5A5A5A5A5A
      RSI: 0x5A5A5A5A5A5A5A5A
      RDI: 0x5A5A5A5A5A5A5A5A
ERROR: Exception occurred during chipsec.modules.tools.smm.smm_ptr.run(): 'unpack requires a string argument of length 56'

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            0.141
[CHIPSEC] Modules total           1
[CHIPSEC] Modules failed to run   1:
ERROR: chipsec.modules.tools.smm.smm_ptr
[CHIPSEC] Modules passed          0:
[CHIPSEC] Modules information     0:
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   0:
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  0:
[CHIPSEC] Modules with exceptions 1:
ERROR: chipsec.modules.tools.smm.smm_ptr
[CHIPSEC] *****************************************************************
```

ErikBjorge, 2019-07-09:
Are you using a version of the Windows driver from a previous release?  If you are, can you build the driver from the source included in 1.3.7 and verify that the error still happens.  The driver return value passing may have been updated to have caused this issue.

We will also take a look at it here.

dummys, 2019-07-09:
Unfortunately I was already using the 1.3.7 driver which I compiled from source

ErikBjorge, 2019-07-09:
That is good to know.  Are you building from the tip of master or are you using one of the release zip files?  Just want to make sure I am using the same code to reproduce the issue on my side.

ErikBjorge, 2019-07-10:
Used the code from the official release version and did not see the issue.  This is looking like a new issue introduced with some of our more recent changes.  I will look at that next.

dummys, 2019-07-11:
yes I downloaded the master. Later today I can give you the exact commit version if you need it ?

ErikBjorge, 2019-07-11:
I will just pull the latest and see what is broken.  I don't think I need the exact commit.

BrentHoltsclaw, 2019-07-15:
if you get a chance can you see if #621 fixes the issue?

dummys, 2019-07-16:
Yes I can confirm that it successfully fixe the issue.
Thanks

closed 2019-07-16 by dummys