chipsec/chipsec #623: Shouldn't this line contain a trailing whitespace?

0xfede7c8, 2019-07-16:
https://github.com/chipsec/chipsec/blob/master/chipsec/helper/linux/helper.py#L1138

I'm using a fork of chipsec with modifications on my own and getting errors because of not having that space.

0xfede7c8, 2019-07-16:
```
ERROR: Exception occurred during chipsec.modules.tools.uefi.whitelist.run(): 'Command '/usr/local/lib/python2.7/dist-packages/chipsec-1.3.7-py2.7-linux-x86_64.egg/chipsec_tools/compression/bin/TianoCompress -d -o sect00_hcty --uefisect00_hcty.gz' returned non-zero exit status 2'
```

BrentHoltsclaw, 2019-07-16:
Looks like you are right, do you want to put in a pull request or would you like us to do it?

0xfede7c8, 2019-07-16:
I can do it! Thanks for the confirmation.

BrentHoltsclaw, 2019-07-16:
Thanks for the help

0xfede7c8, 2019-07-17:
Should I create a PR for python3-rc also?

closed 2019-08-21 by 0xfede7c8