chipsec/chipsec #636: Why I would approximately need 3 weeks on fuzz_test of CHIPSEC 1.3.7-p0011 version?

yunick1, 2019-07-23:
I had tested fuzz_test of CHIPSEC 1.3.6rc1-p0008 version around 40 minutes. But I have tested fuzz_test of CHIPSEC 1.3.7-p0011 version around 3 weeks. Why I would approximately need 3 weeks on fuzz_test of CHIPSEC 1.3.7-p0011 version? Thanks!

BrentHoltsclaw, 2019-07-23:
Which test are you referencing?  We do not have a generic fuzz_test module.  Also can you share the arguments (if any) for the test.

yunick1, 2019-07-24:
Please refer to my test logs in Version 1.3.6rc1-p0008 and Version 1.3.7-p0011. Thanks!

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.3.6rc1-p0008
[CHIPSEC] Arguments: -m tools.smm.smm_ptr -a fuzz,0x00:0xFF
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 4.9.57-onie+ #1 SMP Tue Aug 28 22:54:53 CST 2018 x86_64
[CHIPSEC] Platform: Intel Xeon Processor E5/E7 v4 (Broadwell Server CPU / Wellsburg PCH)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 6F00
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 8C54
 
[+] loaded chipsec.modules.tools.smm.smm_ptr
[*] running loaded modules ..

[*] running module: chipsec.modules.tools.smm.smm_ptr
[*] Module arguments (2):
['fuzz', '0x00:0xFF']
[x][ =======================================================================
[x][ Module: A tool to test SMI handlers for pointer validation vulnerabilies
[x][ =======================================================================
Usage: chipsec_main -m tools.smm.smm_ptr [ -a <mode>,<config_file>|<smic_start:smic_end>,<size>,<address> ]
  mode          SMI handlers testing mode
    = config    use SMI configuration file <config_file>
    = fuzz      fuzz all SMI handlers with code in the range <smic_start:smic_end>
    = fuzzmore  fuzz mode + pass '2nd-order' pointers within buffer to SMI handlers
  size          size of the memory buffer (in Hex)
  address       physical address of memory buffer to pass in GP regs to SMI handlers (in Hex)
    = smram     pass address of SMRAM base (system may hang in this mode!)

[*] Allocated memory buffer (to pass to SMI handlers)       : 0x0000000275D6D000
[*] Allocated 2nd buffer (address will be in the 1st buffer): 0x0000000275D6D800

[*] Configuration
    SMI testing mode          : fuzz
    Range of SMI codes (B2)   : 0x00:0xFF
    Memory buffer pointer     : 0x0000000275D6D000 (address passed in GP regs to SMI)
    Filling/checking memory?  : YES
      Second buffer pointer   : 0x0000000275D6D800 (address written to memory buffer)
      Number of bytes to fill : 0x500
      Byte to fill with       : 0x11
    Additional options (can be changed in the source code):
      Fuzzing SMI functions in ECX?          : 1
      Max value of SMI function in ECX       : 0x10
      Max value of SMI data (B3)             : 0x100
      Max offset of the pointer in the buffer: 0x20
      Passing pointer in all GP registers?   : 0
      Default values of the registers        : 0x5A5A5A5A5A5A5A5A
      Dump all register values every SMI     : 1
      Bail on first detection                : 1

[*] >>> Fuzzing SMI handlers..
[*] AX in RAX will be overwridden with values of SW SMI ports 0xB2/0xB3
    DX in RDX will be overwridden with value 0x00B2

[*] Setting values of general purpose registers to 0x5A5A5A5A5A5A5A5A

[*] reloading buffer with PTR at offset 0x0..
[*] writing 0x500 bytes at 0x0000000275D6D000

[*] fuzzing SMI# 0x00 (data: 0x00)
 >> function (RCX): 0x0000000000000000
    RBX: 0x0000000275D6D000
[... 657 lines cut ...]
      RAX: 0x5A5A5A5A5A5A5A5A
      RBX: 0x0000000178B68800
      RCX: 0x000000000000000A
      RDX: 0x5A5A5A5A5A5A5A5A
      RSI: 0x5A5A5A5A5A5A5A5A
      RDI: 0x5A5A5A5A5A5A5A5A
    < checking buffers
    RSI: 0x0000000178B68800
    > SMI 00 (data: 00)
      RAX: 0x5A5A5A5A5A5A5A5A
      RBX: 0x5A5A5A5A5A5A5A5A
      RCX: 0x000000000000000A
      RDX: 0x5A5A5A5A5A5A5A5A
      RSI: 0x0000000178B68800
      RDI: 0x5A5A5A5A5A5A5A5A
    < checking buffers
    RDI: 0x0000000178B68800
    > SMI 00 (data: 00)
      RAX: 0x5A5A5A5A5A5A5A5A
      RBX: 0x5A5A5A5A5A5A5A5A
      RCX: 0x000000000000000A
      RDX: 0x5A5A5A5A5A5A5A5A
      RSI: 0x5A5A5A5A5A5A5A5A
      RDI: 0x0000000178B68800
    < checking buffers

BrentHoltsclaw, 2019-07-24:
Thank you for the logs.  Looking into the issue.

BrentHoltsclaw, 2019-07-29:
At this point I have not been able to reproduce your failure.  Is it consistently reproducible on your end?  Can you tell more about the system configuration (number of processors/threads, etc)

yunick1, 2019-07-30:
Yes, it is consistently reproducible on our end. We use a processor **(Intel Pentium D1517; 4 Cores; 8 Threads)** in the system configuration. I also attach my system information. You can refer to the following system information. Thanks for your help.

**System information:
	AMI BIOS Core: Aptio V
	Bios Label: 5.011_Grangeville_0ACFK_032
	Intel platform: Broadwell-DE (Grangeville) V2,V3, Intel® Pentium® Processor D1517
	Chipsec version: 1.3.7-p0011**

yunick1, 2019-08-16:
Do you have any update? Thanks!

BrentHoltsclaw, 2019-08-16:
The only changes in the code that I found between 1.3.6 and 1.3.7 which seem to be relevant is the change in IOCTL.  The test module, helper, and driver did not change.  

Does 1.3.8 produce a similar slowdown?

Sae86, 2022-02-16:
Closing due to lack of response, Please contact us at chipsec@intel.com for further questions.

closed 2022-02-16 by Sae86