chipsec/chipsec #643: not an issue, rather a question about BIOSWE and BLE

dummys, 2019-07-29:
Using  chipsec I was able to dump the value of those:

    [00] BIOSWE           = 0 << BIOS Write Enable
    [01] BLE              = 1 << BIOS Lock Enable

My question is: If I unsolder the flash, and reprogram it to disable BLE, is it possible ? I have trouble understanding exactly what is what inside the flash dump. Do you have some offset on where to change this bit ? Thanks

skochinsky, 2019-07-29:
these flags are not in the flash image but in the hardware registers on your PC. they are set by the BIOS, so you would need to find and patch out the code which does it. Check [this course](http://www.opensecuritytraining.info/IntroBIOS.html), it covers some of this topic (in _SPI Flash Protection Mechanisms_).

dummys, 2019-07-29:
nice read I will look at it. But when the computer is off, where it is stored ? Not inside the SPI flash ?

skochinsky, 2019-07-29:
it's not stored anywhere, it must be set on every boot by the BIOS.

dummys, 2019-07-29:
ok got it, thanks. So the only way to attack this is to overwrite the uefi code which set the bits

BrentHoltsclaw, 2019-10-10:
Issue appears to be answered.  Please reopen if farther clarification is needed.

closed 2019-10-10 by BrentHoltsclaw