chipsec/chipsec #644: win7\driver.c(1208): error C2220: warning treated as error - no 'object' file generated

zirkaiva, 2019-07-29:
Trying to build the Windows driver. Buld fails with the following errors:
chipsec\drivers\win7\driver.c(1208): warning C4054: 'type cast': from function pointer 'CPU_REG_TYPE 
chipsec\drivers\win7\driver.c(1208): error C2220: warning treated as error - no 'object' file generated
Offending code:
`                result = hypercall(regs[0], regs[1], regs[2], regs[3], regs[4], regs[5], regs[6], regs[7], regs[8], regs[9], regs[10], (void*)&hypercall_page);
`

BrentHoltsclaw, 2019-07-29:
Which build method and compiler version is giving this problem?

zirkaiva, 2019-07-29:
Building CHIPSEC kernel driver with Visual Studio 2017

BrentHoltsclaw, 2019-07-30:
Sorry, should have also asked where you are building from (release or branch/tag).

zirkaiva, 2019-07-30:
I cloned the repository branch in VS2017

BrentHoltsclaw, 2019-07-30:
Can you check to see if Pull Request #646 fixes the issue?

zirkaiva, 2019-08-01:
Now I  got this error:

`1>------ Build started: Project: chipsec_hlpr, Configuration: Debug|Win32 ------
  Building 'chipsec_hlpr' with toolset 'WindowsKernelModeDriver10.0' and the 'Universal' target platform.
  Stamping Debug\chipsec_hlpr.inf
  Stamping [Version] section with DriverVer=08/01/2019,10.15.37.138
  Assembling amd64\cpu.asm...
  Assembling i386\cpu.asm...
  driver.c
  chipsec_hlpr.vcxproj -> C:\Users\user\Source\Repos\chipsec2\drivers\win7\Debug\chipsec_hlpr.sys
C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1370,5): error MSB4018: The "SignTask" task failed unexpectedly.
C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1370,5): error MSB4018: System.Security.Cryptography.CryptographicException: Access is denied.
C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1370,5): error MSB4018: 
C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1370,5): error MSB4018:    at Microsoft.DriverKit.Build.Shared.CreateCertificate.CryptAcquireContext(String keyContainerName, String providerName, PROV providerType)
C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1370,5): error MSB4018:    at Microsoft.DriverKit.Build.Shared.CreateCertificate.CreateSelfSignedCertificate()
C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1370,5): error MSB4018:    at Microsoft.DriverKit.Build.Tasks.SignTask.AutoGenerateTestCertificate()
C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1370,5): error MSB4018:    at Microsoft.DriverKit.Build.Tasks.SignTask.AutoPickTestCertificate()
C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1370,5): error MSB4018:    at Microsoft.DriverKit.Build.Tasks.SignTask.GenerateCommandLineCommands()
C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1370,5): error MSB4018:    at Microsoft.Build.Utilities.ToolTask.Execute()
C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1370,5): error MSB4018:    at Microsoft.DriverKit.Build.Tasks.TrackedVCToolTask.Execute()
C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1370,5): error MSB4018:    at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute()
C:\Program Files (x86)\Windows Kits\10\build\WindowsDriver.common.targets(1370,5): error MSB4018:    at Microsoft.Build.BackEnd.TaskBuilder.<ExecuteInstantiatedTask>d__26.MoveNext()
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========`

BrentHoltsclaw, 2019-08-01:
Glad it's working.  I'll add a documentation update to try running the build in administrative mode as well if the driver signing is failing.

BrentHoltsclaw, 2019-10-10:
Issue appears to be resolved.  Please reopen if needed.

closed 2019-10-10 by BrentHoltsclaw