chipsec/chipsec #662: setup error

0xfede7c8, 2019-08-21:
Syntax errors when trying to install from master 1.4.0
`sudo python setup.py install`
`sudo python3 setup.py install`

```python
Extracting chipsec-1.4.0-py3.7-linux-x86_64.egg to /usr/local/lib/python3.7/dist-packages
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.0-py3.7-linux-x86_64.egg/chipsec/modules/tools/secureboot/te.py", line 59
    from __future__ import print_function
                                        ^
SyntaxError: from __future__ imports must occur at the beginning of the file

  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.0-py3.7-linux-x86_64.egg/chipsec/modules/tools/smm/smm_ptr.py", line 300
    if FUZZ_BAIL_ON_1ST_DETECT: raise BadSMIDetected, msg
                                                    ^
SyntaxError: invalid syntax
```

0xfede7c8, 2019-08-21:
The first one is "syntax"
The second one is because python2 -> python3

BrentHoltsclaw, 2019-10-08:
This issue appears to be fixed within python 1.4.2.  Please reopen if the issue persists.

closed 2019-10-08 by BrentHoltsclaw