chipsec/chipsec #680: Failure to install when using python2

hughsie, 2019-09-09:
Using python2 (the default) I see:

    ./setup.py install
    Extracting chipsec-1.4.1-py2.7-linux-x86_64.egg to /usr/lib64/python2.7/site-packages
      File "/usr/lib64/python2.7/site-packages/chipsec-1.4.1-py2.7-linux-x86_64.egg/chipsec/modules/tools/secureboot/te.py", line 68
        print ("{:02X} ".format(ord(s[c])),end=' ')
    SyntaxError: invalid syntax

This of course works with Python 3. Maybe it's time to dump py2 completely :)

BrentHoltsclaw, 2019-09-09:
Unfortunately py2 needs to be supported for another year.  The dumpstr function appears to not be used, so I'm going to just delete it.

BrentHoltsclaw, 2019-10-08:
This issue appears to be resolved with chipsec 1.4.2.  Please reopen if the issue persists.

closed 2019-10-08 by BrentHoltsclaw