chipsec/chipsec #681: Cannot uefi decode anything anymore

hughsie, 2019-09-09:
All the UEFI images I try to decode end with this on master:

An exception occurred Failed to decode file: b'Traceback (most recent call last):
  File "/usr/local/bin/chipsec_util", line 11, in <module>
    load_entry_point(\'chipsec==1.4.1\', \'console_scripts\', \'chipsec_util\')()
  File "/usr/local/lib64/python3.7/site-packages/chipsec-1.4.1-py3.7-linux-x86_64.egg/chipsec_util.py", line 184, in main
    return chipsecUtil.main()
  File "/usr/local/lib64/python3.7/site-packages/chipsec-1.4.1-py3.7-linux-x86_64.egg/chipsec_util.py", line 163, in main
    comm.run()
  File "/usr/local/lib64/python3.7/site-packages/chipsec-1.4.1-py3.7-linux-x86_64.egg/chipsec/utilcmd/uefi_cmd.py", line 243, in run
    decode_uefi_region(_uefi, cur_dir, filename, fwtype)
  File "/usr/local/lib64/python3.7/site-packages/chipsec-1.4.1-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 563, in decode_uefi_region
    fwtype = identify_EFI_NVRAM( region_data )
  File "/usr/local/lib64/python3.7/site-packages/chipsec-1.4.1-py3.7-linux-x86_64.egg/chipsec/hal/uefi.py", line 248, in identify_EFI_NVRAM
    (offset, size, hdr) = uefi_platform.EFI_VAR_DICT[ fw_type ][\'func_getnvstore\']( b )
  File "/usr/local/lib64/python3.7/site-packages/chipsec-1.4.1-py3.7-linux-x86_64.egg/chipsec/hal/uefi_platform.py", line 634, in getNVstore_VSS
    return _getNVstore_VSS(nvram_buf, FWType.EFI_FW_TYPE_VSS)
  File "/usr/local/lib64/python3.7/site-packages/chipsec-1.4.1-py3.7-linux-x86_64.egg/chipsec/hal/uefi_platform.py", line 621, in _getNVstore_VSS
    nvram_start = nvram_buf.find( sign )
TypeError: argument should be integer or bytes-like object, not \'str\'

BrentHoltsclaw, 2019-09-09:
Pushed up a small change which should fix the issue.  We need to refactor the code for decompress.  @ErikBjorge what are your thoughts?

frinzell, 2021-02-22:
Closing as fixed.  Please reopen if needed.

closed 2021-02-22 by frinzell