chipsec/chipsec #704: SPI write broken by commit 2efeda4 ("Merge python3-rc2 branch into master (#656)", 2019-08-15)

kudraw, 2019-10-11:
Hello everyone and thanks for this fantastic tool!

I found an issue with the SPI write command, that seems introduced by commit 2efeda4 ("Merge python3-rc2 branch into master (#656)", 2019-08-15).

Launching the command: `./chipsec_util.py spi write 0x181000 return.bin` results in the following:
>[CHIPSEC] Executing command 'spi' with args ['write', '0x181000', 'return.bin']
>[CHIPSEC] writing to SPI flash memory at FLA = 0x181000 from 'return.bin                    '
>Traceback (most recent call last):
>  File "./chipsec_util.py", line 188, in <module>
>    sys.exit( main() )
>  File "./chipsec_util.py", line 184, in main
>    return chipsecUtil.main()
>  File "./chipsec_util.py", line 163, in main
>    comm.run()
>  File "/home/giulio/GIT/chipsec/chipsec/utilcmd/spi_cmd.py", line 91, in run
>    ok = _spi.write_spi_from_file( spi_fla, filename )
>  File "/home/giulio/GIT/chipsec/chipsec/hal/spi.py", line 551, in write_spi_from_file
>    return self.write_spi( spi_fla, struct.unpack('c'*len(buf), buf) )
>  File "/home/giulio/GIT/chipsec/chipsec/hal/spi.py", line 616, in write_spi
>    r = data_byte_count.format(dbc)
> AttributeError: 'int' object has no attribute 'format'

The commit changes a legit integer modulo operation with format in the `write_spi` function of `chipsec/hal/spi.py`, I guess some automatic script error:

```diff
@@ -611,10 +612,10 @@ class SPI(hal_base.HALBase):
         write_ok = True
         data_byte_count = len(buf)
         dbc = 4
-        n = data_byte_count / dbc
-        r = data_byte_count % dbc
+        n = data_byte_count // dbc
+        r = data_byte_count.format(dbc)
```

BrentHoltsclaw, 2019-10-14:
Thanks for the bug report.  Feel free to open a pull request in the future.

frinzell, 2021-02-22:
Closing as fixed.  Please reopen if needed.

closed 2021-02-22 by frinzell