chipsec/chipsec #710: Module directory structure

BrentHoltsclaw, 2019-10-18:
Please review [proposed changes](https://github.com/chipsec/chipsec/wiki/Reorganize-Module-Directory-Structure) and provide comments if something is missing or this will impact any projects.

skochinsky, 2019-10-21:
You have the same names in` modules/` and `tools/`, which is it supposed to be?

BrentHoltsclaw, 2019-10-21:
The goal would be to move the tools directory out of the modules directory and for each to share a similar sub-directory structure that better defines the component the module or tool is evaluating.

The modules directory would be used for settings/validation type testing where the test is searching for a specific configuration and passing back results.  

The tools directory would be used for test cases which could be harmful to or modify system configurations and passing back results.

Sae86, 2022-04-19:
Closing this issue since PR was closed from lack of support.

closed 2022-04-19 by Sae86