chipsec/chipsec #712: Can not get uefi var-list from chipsec_util

GeorgeGhost, 2019-10-19:
Here is the output:
`[CHIPSEC] Executing command 'uefi' with args ['var-list']

[CHIPSEC] Enumerating all EFI variables via OS specific EFI Variable API..
[CHIPSEC] Decoding EFI Variables..
Traceback (most recent call last):
  File "/usr/local/bin/chipsec_util", line 11, in <module>
    load_entry_point('chipsec==1.4.2', 'console_scripts', 'chipsec_util')()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.4.2-py2.7-linux-x86_64.egg/chipsec_util.py", line 184, in main
    return chipsecUtil.main()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.4.2-py2.7-linux-x86_64.egg/chipsec_util.py", line 163, in main
    comm.run()
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.4.2-py2.7-linux-x86_64.egg/chipsec/utilcmd/uefi_cmd.py", line 158, in run
    decode_EFI_variables( efi_vars, nvram_pth )
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.4.2-py2.7-linux-x86_64.egg/chipsec/hal/uefi.py", line 233, in decode_EFI_variables
    write_file( var_fname, data )
  File "/usr/local/lib/python2.7/dist-packages/chipsec-1.4.2-py2.7-linux-x86_64.egg/chipsec/file.py", line 80, in write_file
    f.write( buffer )
UnicodeEncodeError: 'ascii' codec can't encode character u'\x80' in position 4: ordinal not in range(128)
`

NicholasArmour, 2019-10-20:
pr #714 should fix this.

GeorgeGhost, 2019-10-20:
relpaced the code and recombiled from @NicholasArmour clone, still shows same issue

NicholasArmour, 2019-10-20:
are you running this on linux? I assumed so based on the paths in your initial error report.

ive got this working on python2 and 3 on my machine with this fix of removing bytestostring() in linuxhelper.py.

GeorgeGhost, 2019-10-20:
Sorry, it did actually work)

closed 2019-10-20 by GeorgeGhost