chipsec/chipsec #717: Different definitions of PWRMBASE for PCH 3xx/CoffeeLake

martingalvan, 2019-10-21:
Hi, I'm doing a comparison between CPUs' .xml config files and their respective PCHs', and noticed that pch_3xx.xml has the following:

```
    <register name="PWRMBASE" type="pcicfg" bus="0" dev="0x1f" fun="2" offset="0x10" size="8" desc="PM Base Address">
      <field name="STYPE" bit="0"  size="1"  desc="Space Type (always 0 - memory space)"/>
      <field name="BA"    bit="13" size="51" desc="Base Address"/>
    </register>
```

while cfl.xml has this:

```
    <register name="PWRMBASE" type="pcicfg" bus="0" dev="0x1f" fun="2" offset="0x48" size="4" desc="PM Base Address">
      <field name="STYPE" bit="0"  size="1"  desc="Space Type (always 0 - memory space)"/>
      <field name="BA"    bit="12" size="20" desc="Base Address"/>
    </register>
```

In addition, I saw that the Intel datasheet for the 300 series PCH seems to have a mistake, since for PWRMBASE the fields are described as follows:

```
Bit Range | Default and Access | Field Name (ID): Description
----------+--------------------+-----------------------------
31        | 0h RO              | Reserved.
30:12     | 0h RW              | BAR (BASEADDR): Software programs this register with the base address of the device's memory region
11:13     | 0h RO              | Reserved.
12:4      | 0h RO              | Size Indicator (SIZEINDICATOR): Hardwired to 0 to indicate 16KB of memory space
...
```

Where did the values for the 300 series' PWRMBASE come from in the xml? I saw that @ErikBjorge did some fixups as part of https://github.com/chipsec/chipsec/pull/448, but I'm not seeing where these values all come from. Notice that I haven't looked at the whole .xml, so there may be more differences.

BrentHoltsclaw, 2019-10-21:
The chipset initialization first loads common.xml.  The processor configuration can override the common configuration, afterwards if a pch is detected, it will override the configuration yet again.  
 
CFL has a Desktop and Mobile verison [datasheets](https://www.intel.com/content/www/us/en/products/docs/processors/core/core-technical-resources.html).  Following the above logic, the mobile configuration is contained within the cfl.xml so it will be loaded on top of common.  If the pch is not detected, then it will use the mobile register definitions.  If the pch is detected to be within the 300 series, the values will override the configuration data and it will use the definitions from the 300 series pch.

martingalvan, 2019-10-21:
Thanks for the response. Yes, I was aware of how this overriding scheme works in Chipsec; however, I don't fully understand how the Desktop and Mobile versions come into play here. In fact, I'm not seeing a definition for PWRMBASE on the Desktop CFL datasheet you just linked, while it is present in the 300 Series PCH one.

I'm seeing that PWRMBASE appears in the I/O spec for the Mobile CFL. Pardon my ignorance, but why does the PCH datasheet apply only to Desktop versions, and the I/O spec applies to Mobile? It's all a bit confusing.

BrentHoltsclaw, 2019-10-21:
Looking into this a little bit more, the 7th/8th gen mobile PCH will be identified as PCH_100 series as they share the same definitions (with #702).  Therefore the CFL.xml will default to mobile if the pch is not found.

martingalvan, 2019-10-21:
I'm also seeing that the table in the Intel site links to the 7th gen datasheets where it says "These documents apply to 8th Generation Intel® Core™ processors i7-8xxxU, i5-8xxxU, and i9-8xxxHK". The I/O links do point to 8th gen datasheets though. Is this correct, or a mistake on the Intel site?

BrentHoltsclaw, 2019-10-21:
The 300 series pch datasheet can be found [here](https://www.intel.com/content/www/us/en/products/docs/chipsets/300-series-chipset-on-package-pch-datasheet-vol-2.html).

martingalvan, 2019-10-21:
Yes, I had seen this link. What I don't understand is why the PCH datasheet applies for Desktop, and the I/O one applies for Mobile. I also don't understand why the non-I/O 8th gen links in the Intel site point to 7th gen datasheets.

martingalvan, 2019-10-22:
@BrentHoltsclaw about https://github.com/chipsec/chipsec/pull/702, I'm a bit confused now. Even if the definitions in kbl.xml and pch_1xx.xml are the same, wouldn't it be more correct to use kbl.xml for PCHs within the 0x9D40-0x9D5F range? After all, those are mentioned within the I/O datasheet for Mobile CPUs.

BrentHoltsclaw, 2019-10-22:
On older platforms a generation of processor and PCH were normally paired together making the platform xml files all that were necessary.  Over time one processor may be paired with different PCH versions, due to different functionality and requirements.  This has led to the need to detect the processor and PCH separately. CHIPSEC has been keeping the configurations within the platform specific files because if a DID for the PCH is not established, then it will still provide results rather than citing that the platform did is not defined or fall back to the common definitions which will provide inaccurate results.  The proper thing to do with 702 would be to remove the PCH specific configuration from kbl.  Just as the proper thing to do with the cfl configuration would be to remove the PCH specific configuration.

martingalvan, 2019-10-23:
Alright, understood. Thanks!

BrentHoltsclaw, 2019-10-24:
For the cfl example, the PCH settings should either come from the pch_3xx.xml settings for Desktop and the pch_1xx.xml for the mobile.  Therefore any of the configuration settings that are specified within cfl.xml which are not specific to the CPU should be removed from the configuration file.
Realistically to make this work another change would need to be made that allows CHIPSEC to know when it should load a PCH configuration file to ensure that users are aware that CHIPSEC did not find a known DID.

martingalvan, 2019-10-24:
Right, but then why are there separate 8th/9th generation I/O datasheets? Even if the registers are the same as those from the 100 series, I imagine there's a different PCH involved.

I'm sorry to keep nagging about this, I just want to make sure I understand it thoroughly.

BrentHoltsclaw, 2019-10-24:
You are correct the 7th/8th gen mobile pch should be within it's own file.  Because all the registers that CHIPSEC cares about validating match the DID were added to the 100 series pch and not to a new 7th/8th gen mobile pch configuration file.

martingalvan, 2019-10-28:
Alright. Closing this issue.

closed 2019-10-28 by martingalvan