chipsec/chipsec #723: UEFI decode broken

0xfede7c8, 2019-10-24:
[X441SAAS.zip](https://github.com/chipsec/chipsec/files/3769134/X441SAAS.zip)
 For attached file:

```bash
chipsec_util --no_driver uefi decode X441SAAS.303
```

```python
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 369, in build_efi_tree
    fwbin = build_efi_file_tree ( _uefi, fv_img, fwtype)
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 340, in build_efi_file_tree
    fwbin.children = build_efi_modules_tree( _uefi, fwtype, fwbin.Image, fwbin.Size, fwbin.HeaderSize, polarity )
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 266, in build_efi_modules_tree
    sec.ui_string = sec.Image[sec.HeaderSize:-2].decode("utf-16")
UnicodeDecodeError: 'utf-16-le' codec can't decode bytes in position 30-31: illegal UTF-16 surrogate
```
For this one:

[image.zip](https://github.com/chipsec/chipsec/files/3769155/image.zip)

```python
  File "/usr/local/bin/chipsec_util", line 11, in <module>
    load_entry_point('chipsec==1.4.3', 'console_scripts', 'chipsec_util')()
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec_util.py", line 184, in main
    return chipsecUtil.main()
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec_util.py", line 163, in main
    comm.run()
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec/utilcmd/uefi_cmd.py", line 243, in run
    decode_uefi_region(_uefi, cur_dir, filename, fwtype)
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 558, in decode_uefi_region
    parse_uefi_region_from_file( _uefi, fname, fwtype, fv_pth )
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 540, in parse_uefi_region_from_file
    tree = build_efi_model(_uefi, rom, fwtype)
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 425, in build_efi_model
    model = build_efi_tree( _uefi, data, fwtype )
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 369, in build_efi_tree
    fwbin = build_efi_file_tree ( _uefi, fv_img, fwtype)
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 340, in build_efi_file_tree
    fwbin.children = build_efi_modules_tree( _uefi, fwtype, fwbin.Image, fwbin.Size, fwbin.HeaderSize, polarity )
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 304, in build_efi_modules_tree
    sec.children = build_efi_modules_tree( _uefi, fwtype, d, len(d), 0, polarity )
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 313, in build_efi_modules_tree
    children = build_efi_file_tree( _uefi, sec.Image[sec.HeaderSize:], fwtype )
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 350, in build_efi_file_tree
    foff, next_offset, fname, ftype, fattr, fstate, fcsum, fsz, fimg, fhdrsz, fUD, fcalcsum = NextFwFile( fv_img, fv_size, next_offset, polarity )
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.3-py3.7-linux-x86_64.egg/chipsec/hal/uefi_common.py", line 685, in NextFwFile
    fsize = struct.unpack("Q",FvImage[fof+file_header_size:fof+file_header_size+struct.calcsize("Q")])[0]
struct.error: unpack requires a buffer of 8 bytes
```

And there are probably more cases:

BrentHoltsclaw, 2019-10-24:
@0xfede7c8 could you please check #726.  It seems to decompress both images without errors, but can you help validate the correctness of the results.

0xfede7c8, 2019-10-25:
Checked against Chipsec 1.3.6 and your branch reaches the same result for the images provided. I can consider it ok.

closed 2019-12-19 by 0xfede7c8