chipsec/chipsec #729: Wrong definition for ACPI Base Address register in 300 series PCH?

martingalvan, 2019-10-28:
Hi, I'm looking at pch_3xx.xml and saw the following:

    <register name="ABASE" type="pcicfg" bus="0" dev="0x1f" fun="2" offset="0x40" size="4" desc="ACPI Base Address">
      <field name="STYPE" bit="0" size="1" desc="Space Type (always 1 - I/O space)"/>
      <field name="BA"    bit="8" size="8" desc="Base Address"/>
    </register>

However, the Intel 300 and C240 series PCH datasheet, vol 2 shows a different definition for this register (which it calls `ACPIBA`). Here it's described as a MSG Register at offset 0x27B4, with a 14-bit wide `ADDR` field starting from bit 2. The "On-Package" PCH datasheet shows the same.

Is this the same register? If so, why are the definitions different?

kc8apf, 2019-10-30:
As far as I know, my Whiskey Lake system has a 300 series PCH.  I can test an updated configuration.

BrentHoltsclaw, 2019-10-30:
Ok didn't understand that 3xx was broken.  Given the differences within the datasheets, it does not appear that a new configuration file needs to be created at this time.  Could you check #736.

martingalvan, 2019-10-30:
Well, the only relevant difference I saw between the on-package and regular versions is that of https://github.com/chipsec/chipsec/issues/730. I'll check #736 now, thanks!

kc8apf, 2019-10-30:
My Whiskey Lake system reports as follows:
> [CHIPSEC] API mode: using CHIPSEC kernel module API
> [CHIPSEC] OS      : Windows 10 10.0.19008 AMD64
> [CHIPSEC] Python  : 2.7.17 (64-bit)
> [CHIPSEC] Platform: Mobile 8th Generation Core Processor (Whiskey Lake U 4 Cores)
> [CHIPSEC]      VID: 8086
> [CHIPSEC]      DID: 3E34
> [CHIPSEC]      RID: 0B
> [CHIPSEC] PCH     : Intel 300 series On-Package PCH
> [CHIPSEC]      VID: 8086
> [CHIPSEC]      DID: 9D84
> [CHIPSEC]      RID: 30

I'm willing to run tests if someone describes what to run.

BrentHoltsclaw, 2019-10-31:
@kc8apf With the current version of chipsec and with #736.  If you could run `chipsec_util.py io list` and `chipsec_util.py mmio list`.

kc8apf, 2019-10-31:
```
>python chipsec_util.py io list

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.4.3
[CHIPSEC] OS      : Windows 10 10.0.19008 AMD64
[CHIPSEC] Python  : 2.7.17 (64-bit)

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Platform: Mobile 8th Generation Core Processor (Whiskey Lake U 4 Cores)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 3E34
[CHIPSEC]      RID: 0B
[CHIPSEC] PCH     : Intel 300 series On-Package PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 9D84
[CHIPSEC]      RID: 30
[CHIPSEC] Executing command 'io' with args ['list']

--------------------------------------------------------------------------------
 I/O Range    | BAR Register   | Base             | Size     | En? | Description
--------------------------------------------------------------------------------
 PMBASE       | PMBASE         | 0000000000000000 | 00000100 | 1   | ACPI Base Address
 SMBUS_BASE   | 00:1F.4 + 0x20 | 000000000000EFA0 | 00000080 | 1   | SMBus Base Address
 GPIOBASE     | GPIOBASE       | 0000000000000000 | 00000080 | 1   | GPIO Base Address
 TCOBASE      | TCOBASE        | 0000000000000400 | 00000020 | 1   | TCO Base Address
 ABASE        | ABASE          | 0000000000000000 | 00000100 | 1   | ACPI Base Address
```

```
>python chipsec_util.py mmio list

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.4.3
[CHIPSEC] OS      : Windows 10 10.0.19008 AMD64
[CHIPSEC] Python  : 2.7.17 (64-bit)

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Platform: Mobile 8th Generation Core Processor (Whiskey Lake U 4 Cores)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 3E34
[CHIPSEC]      RID: 0B
[CHIPSEC] PCH     : Intel 300 series On-Package PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 9D84
[CHIPSEC]      RID: 30
[CHIPSEC] Executing command 'mmio' with args ['list']

--------------------------------------------------------------------------------
 MMIO Range   | BAR Register   | Base             | Size     | En? | Description
--------------------------------------------------------------------------------
 GTTMMADR     | 00:02.0 + 0x10 | 000000604A000000 | 00001000 | 1   | Graphics Translation Table Range
 GFXVTBAR     | GFXVTBAR       | 00000000FED90000 | 00001000 | 1   | Intel Processor Graphics VT-d Register Range
 SPIBAR       | 00:1F.5 + 0x10 | 00000000FE010000 | 00001000 | 1   | SPI Controller Register Range
 HDABAR       | 00:1F.3 + 0x10 | 0000007FFFEFC000 | 00001000 | 1   | HD Audio Base
 PWRMBASE     | PWRMBASE       | 00000000FE000000 | 00001000 | 1   | Power Management Register Range
 GMADR        | 00:02.0 + 0x18 | 0000004000000000 | 00001000 | 1   | Graphics Memory Range
 DMIBAR       | 00:00.0 + 0x68 | 00000000FED18000 | 00001000 | 1   | Root Complex Register Range
 MMCFG        | 00:00.0 + 0x60 | 00000000E0000000 | 00001000 | 1   | PCI Express Register Range
 RCBA         | 00:1F.0 + 0xF0 | 0000000000000000 | 00004000 | 0   | PCH Root Complex Register Range
 VTBAR        | VTBAR          | 00000000FED91000 | 00001000 | 1   | Intel VT-d Register Register Range
 MCHBAR       | 00:00.0 + 0x48 | 00000000FED10000 | 00008000 | 1   | Host Memory Mapped Register Range
 PXPEPBAR     | 00:00.0 + 0x40 | 00000000FED19000 | 00001000 | 1   | PCI Express Egress Port Register Range
 RCBA_RTC     | 00:1F.0 + 0xF0 | 0000000000003400 | 00000200 | 0   | General Control Register Range
 SBREGBAR     | SBREG_BAR      | 00000000FD000000 | 01000000 | 1   | Sideband Register Access BAR
 HDBAR        | 00:1B.0 + 0x10 | 0000007FFFFFC000 | 00001000 | 1   | PCH HD Audio Controller Register Range
[CHIPSEC] (mmio) time elapsed 0.008
```

BrentHoltsclaw, 2019-11-01:
@martingalvan based upon the results, it appears that the registers are being hidden by hardware.  I agree that we should have them documented correctly, but we really need to ensure that the fixed address is correct for the registers.

martingalvan, 2019-11-01:
Well, AFAIK these fixed addresses come from reference code so I don't know how to help there. However, wouldn't the datasheet be enough to confirm the ABASE and PMBASE changes? If not, how do we normally test the definitions of such hidden registers (e.g. P2SBC)?

ErikBjorge, 2019-11-01:
For ACPI/PMBASE I normally write to the Sleep Type field and then set the Sleep Enable bit to send the system to S5.  This is normally done after verifying that a device is actually decoding the I/O transaction.  This allows me to write a known value to a register and verify that the system performs the expected action.

Similar operations can be done for other register spaces.

martingalvan, 2019-11-04:
> This is normally done after verifying that a device is actually decoding the I/O transaction. 

Pardon my ignorance, but how do you verify this?

ErikBjorge, 2019-11-04:
If no device is decoding an I/O or MMIO region it will return 0xFF for the entire range.  If you get something other than 0xFF for the range you expect you can then try to change settings and verify the expected behavior.  My example was to modify the Sleep Type and set the Sleep Enable values to send the system to S5.  This is easy to verify.

This is not a perfect solution but is better than hoping you have the right values.

martingalvan, 2019-11-06:
Understood.

My main concern here is the `SMI_EN` register, which is used in the `bios_smi` module. For the 300 series, this register is described in the "PMC I/O Based Registers Summary" section:

> The ACPI power management I/O registers are accessed based upon offsets from PM Base Address, BAR2, defined in PCI Device 31: Function 2.

This `BAR 2` register is defined at offset 0x20. Confusingly enough, both `PMBASEA` and `PWRMBASE` are also described as "PM Base Address" in different parts of the datasheet. As for `PMBASEA`, I'm not seeing anything that seems to use it as a BAR. The same goes for `ABASE`.

To summarize: in 3xx, PMC IO registers are based off `BAR 2`, while MMIOs are based off `PWRMBASE`. I'm not seeing stuff based on either `PMBASEA` or `ABASE`, so I'm not sure how to test those (the S5 mode test @ErikBjorge mentioned would need the value of `BAR 2`).

On a related note, we should define `BAR 2` and change the `SMI_EN` definition in 3xx.xml. I'll reference this in #736.

Sae86, 2022-05-04:
This issue seems to have been resolved with #853, closing as fixed but please reopen if needed

closed 2022-05-04 by Sae86