chipsec/chipsec #730: Possible mistake in definition for PWRMBASE for 300 series on-package PCH

martingalvan, 2019-10-28:
Hi, I'm looking at pch_3xx.xml and saw the following definition for `PWRMBASE`:

```
    <register name="PWRMBASE" type="pcicfg" bus="0" dev="0x1f" fun="2" offset="0x10" size="8" desc="PM Base Address">
      <field name="STYPE" bit="0"  size="1"  desc="Space Type (always 0 - memory space)"/>
      <field name="BA"    bit="13" size="51" desc="Base Address"/>
    </register>
```

The Intel 300 and C240 series PCH datasheet, vol 2 shows that there are two registers called `PWRMBASE` and `PWRMBASE HIGH`. I understand that the idea here is to merge them both into one, and the `BA` field will span from bit 13 of `PWRMBASE` to bit 31 of `PWRMBASE HIGH`. This works fine for "regular" 300 series PCHs, however, it appears that there's also an "on-package" 300 series PCH with its own datasheet ([here](https://www.intel.com/content/www/us/en/products/docs/chipsets/300-series-chipset-on-package-pch-datasheet-vol-2.html)). According to this datasheet, the base address field of `PWRMBASE` actually starts at bit 12, and bit 31 is reserved.

The reason I'm reporting this is that chipset.py lists this on-package PCH (DID 0x9D84) as part of the regular 300 series, which means that pch_3xx.xml will get picked for it. I think there should be specific xmls for these on-package PCHs.

BrentHoltsclaw, 2019-10-29:
Good find.  It appears that a pch_3xx-lp.xml will need to be created with the appropriate registers.

Sae86, 2022-05-04:
This issue (and mentioned issue) seem to have been resolved with #853, closing as fixed but please reopen if needed

closed 2022-05-04 by Sae86