chipsec/chipsec #731: 16-bit shift when accessing MSG registers?

martingalvan, 2019-10-28:
Hi,

In `mm_msgbus_reg_read` I'm seeing the following:

`reg_val = self.cs.mmio.read_MMIO_reg_dword(mmio_addr, ((port & 0xFF) << 16) | (register & 0xFFFF))`

Where does that 16-bit shift come from? I've seen this in many other applications dealing with this; however, the Intel datasheets say that these registers should be accessed via SBREG_BAR + PortID + Register Offset.

ErikBjorge, 2019-10-29:
Yes, this is one of the things that is not clear in the documentation.  The main point is that you are creating an MMIO address when accessing these registers.  Bits 23:16 are the port ID (device selector) and bits 15:0 are the offset.  This breaks the MMIO space for the device into nice blocks but this is not explained in the documentation.

The base address is normally programmed by the reference code to be 0xFD000000 leaving the high portion of the register set to zero.  We should update the register definition to be 8 bytes to read both the low and high address.  The RBA field definition will also need to be updated.

martingalvan, 2019-10-29:
Hi @ErikBjorge, thanks for the response. @BrentHoltsclaw also mentioned this reference code in #734; is it publicly available?

closed 2019-10-29 by martingalvan