chipsec/chipsec #734: Source for fixed_address BAR attribute?

martingalvan, 2019-10-29:
Hi all, I've seen that some BARs now have a `fixed_address` attribute which is used in case that trying to read a BAR returns all ones. I'm curious as to where this `fixed_address` comes from, and how reliable it is to use it instead of the BAR's actual value.

Thanks!

BrentHoltsclaw, 2019-10-29:
The value is derived from reference code.

martingalvan, 2019-10-29:
Alright. Is this reference code publicly accessible (from a reliable source)?

Thanks.

ErikBjorge, 2019-10-29:
In general the reference code is not publicly available.  However you may be able to get some information from the edk2-platforms repo.  Check out the devel-MinPlatform branch and look in the Silicon directory.  It may have some of the information you need.

https://github.com/tianocore/edk2-platforms/tree/devel-MinPlatform

martingalvan, 2019-10-29:
Awesome. Thanks!

closed 2019-10-29 by martingalvan