chipsec/chipsec #751: I couldn't run chipsec_main and chipsec_util test in [CHIPSEC] Version : 1.4.2-p0018.

yunick1, 2019-11-05:
**System information:
	AMI BIOS Core: Aptio V
	Bios Label: 5.011_Grangeville_0ACFK_032
	Intel platform: Broadwell-DE (Grangeville) V2,V3, Intel® Pentium® Processor D1517
	Chipsec version: 1.4.2-p0018**

Coucld you please help me check the following running issues logs for chipsec_main and chipsec_util test of [CHIPSEC] Version : 1.4.2-p0018? I need your support to solve the issues. Thanks for your help.

yunick1, 2019-11-05:
<<Chipsec_main test>>

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.4.2-p0018
[CHIPSEC] Arguments: 
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 4.9.57-onie+ #1 SMP Mon Nov 4 16:18:44 CST 2019 x86_64
[CHIPSEC] Python  : 2.7.5
[CHIPSEC] Platform: Intel Xeon Processor E5/E7 v4 (Broadwell Server CPU / Wellsburg PCH)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 6F00
[CHIPSEC]      RID: 03
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 8C54
[CHIPSEC]      RID: 05
 
[*] loading common modules from ".lib.python2.7.site-packages.chipsec-1.4.2-py2.7-linux-x86_64.egg.chipsec.modules.common" ..
[+] loaded chipsec.modules.common.sgx_check
[+] loaded chipsec.modules.common.dmirc
[+] loaded chipsec.modules.common.bios_wp
[+] loaded chipsec.modules.common.spi_vcl
[+] loaded chipsec.modules.common.spi_access
[+] loaded chipsec.modules.common.spd_config
[+] loaded chipsec.modules.common.bios_smi
[+] loaded chipsec.modules.common.me_mfg_mode
[+] loaded chipsec.modules.common.smrr
[+] loaded chipsec.modules.common.tclockdn
[+] loaded chipsec.modules.common.srdl
[+] loaded chipsec.modules.common.bios_kbrd_buffer
[+] loaded chipsec.modules.common.rowhammer
[+] loaded chipsec.modules.common.rcba
[+] loaded chipsec.modules.common.spd_wd
[+] loaded chipsec.modules.common.smm
[+] loaded chipsec.modules.common.rtclock
[+] loaded chipsec.modules.common.acpi_lock
[+] loaded chipsec.modules.common.memlock
[+] loaded chipsec.modules.common.feature_lock
[+] loaded chipsec.modules.common.ia32cfg
[+] loaded chipsec.modules.common.tcoctl_lock
[+] loaded chipsec.modules.common.spi_desc
[+] loaded chipsec.modules.common.bios_ts
[+] loaded chipsec.modules.common.iommu
[+] loaded chipsec.modules.common.gpio
[+] loaded chipsec.modules.common.spi_fdopss
[+] loaded chipsec.modules.common.smi_gpio
[+] loaded chipsec.modules.common.ltdpr
[+] loaded chipsec.modules.common.miscctrlsts0
[+] loaded chipsec.modules.common.spi_lock
[+] loaded chipsec.modules.common.errinjcon
[+] loaded chipsec.modules.common.secureboot.variables
[+] loaded chipsec.modules.common.uefi.access_platform
[+] loaded chipsec.modules.common.uefi.access_uefispec
[+] loaded chipsec.modules.common.uefi.s3bootscript
[+] loaded chipsec.modules.common.cpu.ia_untrusted
[+] loaded chipsec.modules.common.cpu.spectre_v2
[*] No platform specific modules to load
[*] loading modules from ".lib.python2.7.site-packages.chipsec-1.4.2-py2.7-linux-x86_64.egg.chipsec.modules" ..
[+] loaded chipsec.modules.debugenabled
[+] loaded chipsec.modules.remap
[+] loaded chipsec.modules.spec_ctrl
[+] loaded chipsec.modules.memconfig
[+] loaded chipsec.modules.smm_dma
[+] loaded chipsec.modules.spd
[+] loaded chipsec.modules.dci
[... 594 lines cut ...]

[*] running module: chipsec.modules.spd
Skipping module chipsec.modules.spd since it is not supported in this platform

[*] running module: chipsec.modules.dci
Skipping module chipsec.modules.dci since it is not supported in this platform

[*] running module: chipsec.modules.configlock
[x][ =======================================================================
[x][ Module: CPU Configuration Lock Check
[x][ =======================================================================
[*] checking that the CPU configuration is locked down by the firmware..
[*] LT_LOCK_MEMORY = 0x       9 << Config Lock MSR (MSR 0x2E7)
    [00] ConfigLock       = 1 << ConfigLock 
[+] PASSED: CPU configuration is locked down

[*] running module: chipsec.modules.mclock
Skipping module chipsec.modules.mclock since it is not supported in this platform

[*] running module: chipsec.modules.biosguard
[x][ =======================================================================
[x][ Module: Intel BIOS Guard Configuration
[x][ =======================================================================
[*] Checking if CPU BIOS Guard is enabled and locked in PLAT_FRMW_PROT_CTRL MSR..
Segmentation fault

yunick1, 2019-11-05:
<<chipsec_util test>>

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.4.2-p0018
[CHIPSEC] OS      : Linux 4.9.57-onie+ #1 SMP Mon Nov 4 16:18:44 CST 2019 x86_64
[CHIPSEC] Python  : 2.7.5
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Platform: Intel Xeon Processor E5/E7 v4 (Broadwell Server CPU / Wellsburg PCH)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 6F00
[CHIPSEC]      RID: 03
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 8C54
[CHIPSEC]      RID: 05
[CHIPSEC] Executing command 'uefi' with args ['var-list']

[CHIPSEC] Enumerating all EFI variables via OS specific EFI Variable API..
[CHIPSEC] Decoding EFI Variables..
Traceback (most recent call last):
  File "/bin/chipsec_util", line 11, in <module>
    load_entry_point('chipsec==1.4.2', 'console_scripts', 'chipsec_util')()
  File "/lib/python2.7/site-packages/chipsec-1.4.2-py2.7-linux-x86_64.egg/chipsec_util.py", line 184, in main
    return chipsecUtil.main()
  File "/lib/python2.7/site-packages/chipsec-1.4.2-py2.7-linux-x86_64.egg/chipsec_util.py", line 163, in main
    comm.run()
  File "/lib/python2.7/site-packages/chipsec-1.4.2-py2.7-linux-x86_64.egg/chipsec/utilcmd/uefi_cmd.py", line 158, in run
    decode_EFI_variables( efi_vars, nvram_pth )
  File "/lib/python2.7/site-packages/chipsec-1.4.2-py2.7-linux-x86_64.egg/chipsec/hal/uefi.py", line 233, in decode_EFI_variables
    write_file( var_fname, data )
  File "/lib/python2.7/site-packages/chipsec-1.4.2-py2.7-linux-x86_64.egg/chipsec/file.py", line 80, in write_file
    f.write( buffer )
UnicodeEncodeError: 'ascii' codec can't encode character u'\x88' in position 108: ordinal not in range(128)

ErikBjorge, 2019-11-05:
Can you please send mail to chipsec@intel.com to help us better handle these failures?

ErikBjorge, 2019-11-06:
Handling over email.

closed 2019-11-06 by ErikBjorge