chipsec/chipsec #755: There are two failures and one error of chipsec_main test in [CHIPSEC] Version : 1.4.2-p0018.

yunick1, 2019-11-13:
**System information:
	AMI BIOS Core: Aptio V
	Bios Label: 5.011_Grangeville_0ACFK_032
	Intel platform: Broadwell-DE (Grangeville) V2,V3, Intel® Pentium® Processor D1517
	Chipsec version: 1.4.2-p0018**

I have skipped BIOS Guard to run chipsec_main test in [CHIPSEC] Version : 1.4.2-p0018. I find the following two failures and one error. Could you please help me explain why there were two failures and one error for chipsec_main test of [CHIPSEC] Version : 1.4.2-p0018? I need your support to clarify the two failures and one error. Thanks for your help.

yunick1, 2019-11-13:
<<two failures and one error of chipsec_main test>>
**[-] FAILED: chipsec.modules.common.rcba
[-] FAILED: chipsec.modules.common.spd_wd
ERROR: chipsec.modules.common.feature_lock**

yunick1, 2019-11-13:
<<two failures and one error of chipsec_main test>>
[-] FAILED: chipsec.modules.common.rcba
[-] FAILED: chipsec.modules.common.spd_wd
ERROR: chipsec.modules.common.feature_lock

yunick1, 2019-11-13:
(two failures and one error of chipsec_main test)

**[-] FAILED: chipsec.modules.common.rcba
[-] FAILED: chipsec.modules.common.spd_wd
ERROR: chipsec.modules.common.feature_lock**

yunick1, 2019-11-13:
(chipsec_main test)

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.4.2-p0018
[CHIPSEC] Arguments: 
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : Linux 4.9.57-onie+ #1 SMP Mon Nov 11 21:40:14 CST 2019 x86_64
[CHIPSEC] Python  : 2.7.5
[CHIPSEC] Platform: Intel Xeon Processor E5/E7 v4 (Broadwell Server CPU / Wellsburg PCH)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 6F00
[CHIPSEC]      RID: 03
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 8C54
[CHIPSEC]      RID: 05
 
[*] loading common modules from ".lib.python2.7.site-packages.chipsec-1.4.2-py2.7-linux-x86_64.egg.chipsec.modules.common" ..
[+] loaded chipsec.modules.common.sgx_check
[+] loaded chipsec.modules.common.dmirc
[+] loaded chipsec.modules.common.bios_wp
[+] loaded chipsec.modules.common.spi_vcl
[+] loaded chipsec.modules.common.spi_access
[+] loaded chipsec.modules.common.spd_config
[+] loaded chipsec.modules.common.bios_smi
[+] loaded chipsec.modules.common.me_mfg_mode
[+] loaded chipsec.modules.common.smrr
[+] loaded chipsec.modules.common.tclockdn
[+] loaded chipsec.modules.common.srdl
[+] loaded chipsec.modules.common.bios_kbrd_buffer
[+] loaded chipsec.modules.common.rowhammer
[+] loaded chipsec.modules.common.rcba
[+] loaded chipsec.modules.common.spd_wd
[+] loaded chipsec.modules.common.smm
[+] loaded chipsec.modules.common.rtclock
[+] loaded chipsec.modules.common.acpi_lock
[+] loaded chipsec.modules.common.memlock
[+] loaded chipsec.modules.common.feature_lock
[+] loaded chipsec.modules.common.ia32cfg
[+] loaded chipsec.modules.common.tcoctl_lock
[+] loaded chipsec.modules.common.spi_desc
[+] loaded chipsec.modules.common.bios_ts
[+] loaded chipsec.modules.common.iommu
[+] loaded chipsec.modules.common.gpio
[+] loaded chipsec.modules.common.spi_fdopss
[+] loaded chipsec.modules.common.smi_gpio
[+] loaded chipsec.modules.common.ltdpr
[+] loaded chipsec.modules.common.miscctrlsts0
[+] loaded chipsec.modules.common.spi_lock
[+] loaded chipsec.modules.common.errinjcon
[+] loaded chipsec.modules.common.secureboot.variables
[+] loaded chipsec.modules.common.uefi.access_platform
[+] loaded chipsec.modules.common.uefi.access_uefispec
[+] loaded chipsec.modules.common.uefi.s3bootscript
[+] loaded chipsec.modules.common.cpu.ia_untrusted
[+] loaded chipsec.modules.common.cpu.spectre_v2
[*] No platform specific modules to load
[*] loading modules from ".lib.python2.7.site-packages.chipsec-1.4.2-py2.7-linux-x86_64.egg.chipsec.modules" ..
[+] loaded chipsec.modules.debugenabled
[+] loaded chipsec.modules.remap
[+] loaded chipsec.modules.spec_ctrl
[+] loaded chipsec.modules.memconfig
[+] loaded chipsec.modules.smm_dma
[+] loaded chipsec.modules.spd
[+] loaded chipsec.modules.dci
[... 665 lines cut ...]
[CHIPSEC] Modules with warnings   2:
[!] WARNING: chipsec.modules.common.uefi.s3bootscript
[!] WARNING: chipsec.modules.common.cpu.spectre_v2
[CHIPSEC] Modules not implemented 9:
[*] NOT IMPLEMENTED: chipsec.modules.common.me_mfg_mode
[*] NOT IMPLEMENTED: chipsec.modules.common.rowhammer
[*] NOT IMPLEMENTED: chipsec.modules.common.rtclock
[*] NOT IMPLEMENTED: chipsec.modules.remap
[*] NOT IMPLEMENTED: chipsec.modules.memconfig
[*] NOT IMPLEMENTED: chipsec.modules.smm_dma
[*] NOT IMPLEMENTED: chipsec.modules.spd
[*] NOT IMPLEMENTED: chipsec.modules.dci
[*] NOT IMPLEMENTED: chipsec.modules.biosguard
[CHIPSEC] Modules not applicable  8:
[*] NOT APPLICABLE: chipsec.modules.common.sgx_check
[*] NOT APPLICABLE: chipsec.modules.common.spd_config
[*] NOT APPLICABLE: chipsec.modules.common.smm
[*] NOT APPLICABLE: chipsec.modules.common.acpi_lock
[*] NOT APPLICABLE: chipsec.modules.common.tcoctl_lock
[*] NOT APPLICABLE: chipsec.modules.common.gpio
[*] NOT APPLICABLE: chipsec.modules.common.cpu.ia_untrusted
[*] NOT APPLICABLE: chipsec.modules.mclock
[CHIPSEC] Modules with exceptions 1:
ERROR: chipsec.modules.common.feature_lock
[CHIPSEC] *****************************************************************

BrentHoltsclaw, 2019-11-13:
Can you please send mail to chipsec@intel.com to help us better handle these failures?

BrentHoltsclaw, 2019-11-13:
Handling over email

closed 2019-11-13 by BrentHoltsclaw