chipsec/chipsec #765: IOSF message bus extended address MCRX register bug

calvinbalke13, 2019-11-22:
I found a problem accessing extended addresses on the IOSF-SB message bus on this line:
https://github.com/chipsec/chipsec/blob/4b2dd0dfc5a2f53e087fd54c896bb0961065d244/chipsec/hal/msgbus.py#L119

I'm pretty sure

`mcrx = self.cs.set_register_field( 'MSG_CTRL_REG_EXT', mcrx, 'MESSAGE_ADDRESS_OFFSET_EXT', reg, preserve_field_position=True )`

should be replaced with

`mcrx = self.cs.set_register_field( 'MSG_CTRL_REG_EXT', mcrx, 'MESSAGE_ADDRESS_OFFSET_EXT', (reg>>8), preserve_field_position=True )`

I verified that this fix was functional by reading extended addresses found in a disassembled UEFI firmware.  Previously it was returning all zeros, and after the fix it started returning data.

However, I don't know how preserve_field_position works so there may be a better way to fix it without needing to resort to doing a right shift.

Please let me know if this helps, Thanks.

BrentHoltsclaw, 2019-11-23:
@calvinbalke13 we're looking into the issue

BrentHoltsclaw, 2019-11-25:
Good Catch.  MCRX is used for messages sent to end points that require more than 8 bits for the offset/register. These bits are a direct extension of MCR[15:8]. Your fix for the issue looks good.  Would you like to push up a pull request with the change?

frinzell, 2021-02-22:
Closing as fixed.

closed 2021-02-22 by frinzell