chipsec/chipsec #781: Create a test suite for the UEFI parser

0xfede7c8, 2019-12-11:
The UEFI parser has sufferent a lot of changes in the recent time.

Idea is that future changes in the parser can be easily tested.

Features:

- Easy inclusion of new test cases (images and data) 
- Test cases should be included on the repository for portability (limitations on having binaries on the repo?)

0xfede7c8, 2019-12-11:
You can assign me to this if you agree.

BrentHoltsclaw, 2019-12-11:
Any idea the size of test cases you would like to create.  I'm wondering if it would be better to create a repository opposed to increasing the size of CHIPSEC?

0xfede7c8, 2019-12-11:
The images are between 8-32MB each.
I think the test cases would contain ~25 images at first.
I will be an average of ~400MB

I'm ok with other repo. Maybe a submodule.

BrentHoltsclaw, 2019-12-12:
As discussed offline, I think defining a format/metrics makes sense.  Please let @npmitche know if you would like any help.  

@hughsie Do you have an opinion?

npmitche, 2022-12-15:
Created a private repo with a small set of binary files for testing.

closed 2022-12-15 by npmitche