chipsec/chipsec #792: 'bytes is not a 16-char string' on UEFI decode

0xfede7c8, 2019-12-19:
[CB410.tar.gz](https://github.com/chipsec/chipsec/files/3984709/CB410.tar.gz)

```
chipsec_util --no_driver uefi decode CB410 

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.4.4
[CHIPSEC] OS      : Linux 5.3.0-24-generic #26-Ubuntu SMP Thu Nov 14 01:33:18 UTC 2019 x86_64
[CHIPSEC] Python  : 3.7.5 (64-bit)
[CHIPSEC] API mode: using OS native API (not using CHIPSEC kernel module)
[CHIPSEC] Helper  : LinuxHelper (None)
[CHIPSEC] Platform: Desktop 8th Generation Core Processor (CoffeeLake H 6 Cores)
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 3EC4
[CHIPSEC]      RID: 07
[CHIPSEC] PCH     : Intel CM246 (300 series) PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: A30E
[CHIPSEC]      RID: 10
[CHIPSEC] Executing command 'uefi' with args ['decode', 'CB410']

[CHIPSEC] Parsing EFI volumes from 'CB410'..
Traceback (most recent call last):
  File "/usr/local/bin/chipsec_util", line 11, in <module>
    load_entry_point('chipsec==1.4.4', 'console_scripts', 'chipsec_util')()
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.4-py3.7-linux-x86_64.egg/chipsec_util.py", line 187, in main
    return chipsecUtil.main()
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.4-py3.7-linux-x86_64.egg/chipsec_util.py", line 166, in main
    comm.run()
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.4-py3.7-linux-x86_64.egg/chipsec/utilcmd/uefi_cmd.py", line 243, in run
    decode_uefi_region(_uefi, cur_dir, filename, fwtype)
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.4-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 566, in decode_uefi_region
    parse_uefi_region_from_file( _uefi, fname, fwtype, fv_pth )
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.4-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 548, in parse_uefi_region_from_file
    tree = build_efi_model(_uefi, rom, fwtype)
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.4-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 433, in build_efi_model
    model = build_efi_tree( _uefi, data, fwtype )
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.4-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 377, in build_efi_tree
    fwbin = build_efi_file_tree ( _uefi, fv_img, fwtype)
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.4-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 348, in build_efi_file_tree
    fwbin.children = build_efi_modules_tree( _uefi, fwtype, fwbin.Image, fwbin.Size, fwbin.HeaderSize, polarity )
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.4-py3.7-linux-x86_64.egg/chipsec/hal/spi_uefi.py", line 284, in build_efi_modules_tree
    sec.Guid = guid_str(guid0, guid1, guid2, guid3)
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.4.4-py3.7-linux-x86_64.egg/chipsec/hal/uefi_common.py", line 488, in guid_str
    guid = uuid.UUID(bytes=guid0[::-1] + guid1[::-1] + guid2[::-1] + guid3[:2] + guid3[-6:])
  File "/usr/lib/python3.7/uuid.py", line 169, in __init__
    raise ValueError('bytes is not a 16-char string')
ValueError: bytes is not a 16-char string
```

BrentHoltsclaw, 2019-12-19:
Pushed up a change which should address within #777 

Are you able to decompress with other tools?  The output I'm getting appears to indicate there are problems with the image.

0xfede7c8, 2019-12-19:
I will check.

It may be a malformed image. The question is, should the parser crash like this? Is this expected?

closed 2019-12-20 by 0xfede7c8