chipsec/chipsec #793: 10th-gen ASUS bios image doesn't decompress properly

NicholasArmour, 2019-12-19:
https://www.asus.com/Laptops/ASUS-ZenBook-14-UX434FLC/HelpDesk_BIOS/

Tried with PR#777, found 106 PE's.   Seem to be missing a lot of DXE+SMM drivers still

BrentHoltsclaw, 2019-12-19:
Appears to be a separate issue than what 777 was trying to fix.

NikolajSchlej, 2020-01-21:
That file is not suitable for being parsed as UEFI image directly, because it's in fact an AMI PFAT capsule (see _AMIPFAT magic at offset 0x8).

It can be converted into a proper UEFI image using [this python3 script](https://github.com/platomav/BIOSUtilities/blob/master/AMI%20BIOS%20Guard%20Extractor/AMI_PFAT_Extract.py), result is attached.
[UX434FLCAS.300_unpacked.bin.zip](https://github.com/chipsec/chipsec/files/4092512/UX434FLCAS.300_unpacked.bin.zip)

Sae86, 2022-06-14:
Closing, as it seems it was an unsupported binary.

closed 2022-06-14 by Sae86