chipsec/chipsec #805: Build error while compiling the driver for OS X

startergo, 2020-01-02:
/Users/g5/Downloads/chipsec-master/drivers/osx/chipsec/cpu.asm:405:20: error: unexpected token in argument list
    out %ax, $0x0B2h
                   ^
Command CompileC failed with a nonzero exit code

BrentHoltsclaw, 2020-01-02:
I believe that #806 fixes the issue.  Can you help confirm.

startergo, 2020-01-02:
It builds now. Thanks

startergo, 2020-01-02:
Upon running the script I get this error:
"sudo python chipsec_util.py spi info

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.4.5
[CHIPSEC] OS      : Darwin 19.2.0 Darwin Kernel Version 19.2.0: Sat Nov  9 03:47:04 PST 2019; root:xnu-6153.61.1~20/RELEASE_X86_64 x86_64
[CHIPSEC] Python  : 2.7.16 (64-bit)
[CHIPSEC] API mode: using CHIPSEC kernel module API
ERROR: Unsupported Platform: VID = 0x8086, DID = 0x3406, RID = 0x22
WARNING: *******************************************************************
WARNING: * Unknown platform!
WARNING: * Platform dependent functionality will likely be incorrect
WARNING: * Error Message: "Unsupported Platform: VID = 0x8086, DID = 0x3406, RID = 0x22"
WARNING: *******************************************************************
[CHIPSEC] Helper  : OSXHelper (/Users/g5/Downloads/chipsec-1.4-2.5/chipsec/helper/osx/chipsec.kext)
[CHIPSEC] Platform: UnknownPlatform
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 3406
[CHIPSEC]      RID: 22
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 3A18
[CHIPSEC]      RID: 00
[CHIPSEC] Executing command 'spi' with args ['info']

Traceback (most recent call last):
  File "chipsec_util.py", line 191, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 187, in main
    return chipsecUtil.main()
  File "chipsec_util.py", line 166, in main
    comm.run()
  File "/Users/g5/Downloads/chipsec-1.4-2.5/chipsec/utilcmd/spi_cmd.py", line 69, in run
    _spi = SPI( self.cs )
  File "/Users/g5/Downloads/chipsec-1.4-2.5/chipsec/hal/spi.py", line 190, in __init__
    self.hsfs_off   = int(self.cs.get_register_def("HSFS")['offset'],16)
  File "/Users/g5/Downloads/chipsec-1.4-2.5/chipsec/chipset.py", line 876, in get_register_def
    reg_def = self.Cfg.REGISTERS[reg_name]
KeyError: 'HSFS'"

BrentHoltsclaw, 2020-01-02:
are you running with the -i flag?

startergo, 2020-01-02:
chipsec-1.4-2.5 % sudo python chipsec_util.py spi info -i
Password:

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.4.5
[CHIPSEC] OS      : Darwin 19.2.0 Darwin Kernel Version 19.2.0: Sat Nov  9 03:47:04 PST 2019; root:xnu-6153.61.1~20/RELEASE_X86_64 x86_64
[CHIPSEC] Python  : 2.7.16 (64-bit)
[CHIPSEC] API mode: using CHIPSEC kernel module API
ERROR: Unsupported Platform: VID = 0x8086, DID = 0x3406, RID = 0x22
WARNING: *******************************************************************
WARNING: * Unknown platform!
WARNING: * Platform dependent functionality will likely be incorrect
WARNING: * Error Message: "Unsupported Platform: VID = 0x8086, DID = 0x3406, RID = 0x22"
WARNING: *******************************************************************
[CHIPSEC] Helper  : OSXHelper (/Users/g5/Downloads/chipsec-1.4-2.5/chipsec/helper/osx/chipsec.kext)
[CHIPSEC] Platform: UnknownPlatform
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 3406
[CHIPSEC]      RID: 22
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 3A18
[CHIPSEC]      RID: 00
[CHIPSEC] Executing command 'spi' with args ['info', '-i']

Traceback (most recent call last):
  File "chipsec_util.py", line 191, in <module>
    sys.exit( main() )
  File "chipsec_util.py", line 187, in main
    return chipsecUtil.main()
  File "chipsec_util.py", line 166, in main
    comm.run()
  File "/Users/g5/Downloads/chipsec-1.4-2.5/chipsec/utilcmd/spi_cmd.py", line 69, in run
    _spi = SPI( self.cs )
  File "/Users/g5/Downloads/chipsec-1.4-2.5/chipsec/hal/spi.py", line 190, in __init__
    self.hsfs_off   = int(self.cs.get_register_def("HSFS")['offset'],16)
  File "/Users/g5/Downloads/chipsec-1.4-2.5/chipsec/chipset.py", line 876, in get_register_def
    reg_def = self.Cfg.REGISTERS[reg_name]
KeyError: 'HSFS'

BrentHoltsclaw, 2020-01-03:
CHIPSEC will currently not load a configuration for an unknown chipset as it will not be accurate.  We would need to add support for CPU DID 3406 and PCH 3A18.  

It appears to be an old device Tylersburg cpu and ICH10.   What would you like to use CHIPSEC for other than dumping the spi?

To dump the spi -ICH10 should be supported by common.xml, therefore if you run with `chipsec_util.py -p bdw spi info` it should give back reasonable results for that command.

startergo, 2020-01-03:
> CHIPSEC will currently not load a configuration for an unknown chipset as it will not be accurate. We would need to add support for CPU DID 3406 and PCH 3A18.
>
> It appears to be an old device [...]

This is x5690 Westmere-EP CPU. I am testing a bootloader based on tianocore, but it is not properly recognizing the CPU frequency, microcode revision and the multiplier, so I am trying to debug it to  see what CPU addresses it queries.

startergo, 2020-01-03:
I can see it can read the registers although it identified incorrectly the CPU.

startergo, 2020-01-03:
I could read register 8b, but reading registers 15 and 16 led to kernel panic:
"{"timestamp":"2020-01-02 23:39:33.32 -0500","bug_type":"210","os_version":"Mac OS X 10.15.2 (19C57)"}
{"macOSProcessedStackshotData":"bm8gb24gZGlzayBvciBzbGVlcC93YWtlIGZhaWx1cmUgcGFuaWMgc3RhY2tzaG90IGZvdW5k","macOSPanicString":"panic(cpu 12 caller 0xffffff800066520a): Kernel trap at 0xffffff7f85508a28, type 13=general protection, registers:\nCR0: 0x0000000080010033, CR2: 0x00007fff8b2115e0, CR3: 0x0000001e6cf040cc, CR4: 0x00000000000226e0\nRAX: 0x0000000000000000, RBX: 0x0000000024000000, RCX: 0x0000000000000015, RDX: 0xffffffce49b6bab0\nRSP: 0xffffffce49b6b7f8, RBP: 0xffffffce49b6bb50, RSI: 0xffffffce49b6baa8, RDI: 0x0000000000000015\nR8:  0x00000010f187abe6, R9:  0xffffff8de1b3fca8, R10: 0xffffffce49b6baa8, R11: 0xffffffce49b6bab0\nR12: 0xffffffce49b6bb90, R13: 0xffffff811a6a2100, R14: 0x0000000024000000, R15: 0xffffff81192295f8\nRFL: 0x0000000000010282, RIP: 0xffffff7f85508a28, CS:  0x0000000000000008, SS:  0x0000000000000010\nFault CR2: 0x00007fff8b2115e0, Error code: 0x0000000000000000, Fault CPU: 0xc, PL: 0, VF: 0\n\nBacktrace (CPU 12), Frame : Return Address\n0xffffff8000357220 : 0xffffff800053bb1b mach_kernel : _handle_debugger_trap + 0x47b\n0xffffff8000357270 : 0xffffff80006733e5 mach_kernel : _kdp_i386_trap + 0x155\n0xffffff80003572b0 : 0xffffff8000664e5e mach_kernel : _kernel_trap + 0x4ee\n0xffffff8000357300 : 0xffffff80004e2a40 mach_kernel : _return_from_trap + 0xe0\n0xffffff8000357320 : 0xffffff800053b207 mach_kernel : _DebuggerTrapWithState + 0x17\n0xffffff8000357420 : 0xffffff800053b5eb mach_kernel : _panic_trap_to_debugger + 0x21b\n0xffffff8000357470 : 0xffffff8000cd24f9 mach_kernel : _panic + 0x61\n0xffffff80003574e0 : 0xffffff800066520a mach_kernel : _sync_iss_to_iks + 0x2aa\n0xffffff8000357660 : 0xffffff8000664f08 mach_kernel : _kernel_trap + 0x598\n0xffffff80003576b0 : 0xffffff80004e2a40 mach_kernel : _return_from_trap + 0xe0\n0xffffff80003576d0 : 0xffffff7f85508a28 com.google.chipsec : _ReadMSR + 0x9\n0xffffffce49b6bb50 : 0xffffff80007d6b02 mach_kernel : _spec_ioctl + 0xb2\n0xffffffce49b6bb80 : 0xffffff80007cae30 mach_kernel : _VNOP_IOCTL + 0xf0\n0xffffffce49b6bc00 : 0xffffff80007bc12e mach_kernel : _utf8_normalizestr + 0xd4e\n0xffffffce49b6bdf0 : 0xffffff8000a5f8ab mach_kernel : _fo_ioctl + 0x7b\n0xffffffce49b6be20 : 0xffffff8000ab86f5 mach_kernel : _ioctl + 0x525\n0xffffffce49b6bf40 : 0xffffff8000b9acba mach_kernel : _unix_syscall64 + 0x28a\n0xffffffce49b6bfa0 : 0xffffff80004e3206 mach_kernel : _hndl_unix_scall64 + 0x16\n      Kernel Extensions in backtrace:\n         com.google.chipsec(1.0)[4463A927-47E5-3A0F-95D1-31A9D882C225]@0xffffff7f85508000->0xffffff7f8550efff\n\nBSD process name corresponding to current thread: Python\
Mac OS version:\n19C57\n\nKernel version:\nDarwin Kernel Version 19.2.0: Sat Nov  9 03:47:04 PST 2019; root:xnu-6153.61.1~20\/RELEASE_X86_64\nKernel UUID: C3E7E405-C692-356B-88D3-C30041FD1E72\nKernel slide:     0x0000000000200000\nKernel text base: 0xffffff8000400000\n__HIB  text base: 0xffffff8000300000\nSystem model name: MacPro5,1 (Mac-7BA5B2D9E42DDD94)\nSystem shutdown begun: NO\nPanic diags file available: NO (0xe00002bc)\n\nSystem uptime in nanoseconds: 1967977205020\n\n\n"}"

Is there a way to read in bulk the registers or some other command without going into kernel panic? I used:

chipsec-1.4-2.5 % sudo ./chipsec_util.py msr 0x15
chipsec-1.4-2.5 % sudo ./chipsec_util.py msr 0x16

startergo, 2020-01-03:
How can I use chipsec.hal.msr module?
It does not seem to work.
usage:
>>> read_msr( 0x8B )
>>> write_msr( 0x79, 0x12345678 )
>>> get_IDTR( 0 )
>>> get_GDTR( 0 )
>>> dump_Descriptor_Table( 0, DESCRIPTOR_TABLE_CODE_IDTR )
>>> IDT( 0 )
>>> GDT( 0 )
>>> IDT_all()
>>> GDT_all()
Is it included in the chipsec_util.py? I know I can read msr with chipsec_util.py msr <MSR address>, but how to use read_msr, write_msr, get_IDTR, etc with chipsec_util.py or chipsec.hal.msr

BrentHoltsclaw, 2020-01-03:
```
$ sudo python
>>> from chipsec.chipset import cs
>>> from chipsec.hal.msr import Msr
>>> _cs = cs()
>>> _cs.init(None,None,None)
>>> _msr = MSR(_cs)
>>> _msr.read_msr(0,0x8b)
...
```

startergo, 2020-01-03:
After _cs.init I got errors:
">>> _cs.init(None,None,True)
[CHIPSEC] API mode: using CHIPSEC kernel module API
ERROR: Unsupported Platform: VID = 0x8086, DID = 0x3406, RID = 0x22
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "chipsec/chipset.py", line 563, in init
    raise UnknownChipsetError (msg)
chipsec.chipset.UnknownChipsetError: Unsupported Platform: VID = 0x8086, DID = 0x3406, RID = 0x22
>>> _msr = MSR(_cs)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'MSR' is not defined
>>> _msr.read_msr(0,0x8b)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name '_msr' is not defined"

BrentHoltsclaw, 2020-01-03:
try
`>>> _cs.init("BDW",None,True)`

BrentHoltsclaw, 2020-01-03:
That functionality is a bug, I'll work on a fix.

startergo, 2020-01-03:
BDW worked but:
>>> _cs.init("BDW",None,True)
[CHIPSEC] API mode: using CHIPSEC kernel module API
>>> _msr = MSR(_cs)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'MSR' is not defined
>>> _msr.read_msr(0,0x8b)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name '_msr' is not defined

BrentHoltsclaw, 2020-01-03:
Sorry for the typo
`>>> _msr = Msr(_cs)`

startergo, 2020-01-03:
Your workaround works. Great!

frinzell, 2021-02-22:
Closing as fixed.  Please reopen if needed.

closed 2021-02-22 by frinzell