chipsec/chipsec #810: Unable to find the register (GCS RCBA + 0x3410) for Denverton(DNV) CPU

hans-cameo, 2020-01-06:
The CPU is **Denverton**, Intel Atom® Processor C3708.
I failed at a module “BIOS Interface Lock (including Top Swap Mode)”
Does the register CHIPSEC try to get exists in Denverton CPU?

Here is the log:

CHIPSEC Info:
```
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.4.4
[CHIPSEC] Arguments: -p DNV -l denverton.log
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : uefi   
[CHIPSEC] Python  : 2.7.2 (64-bit)
[CHIPSEC] Helper  : EfiHelper (None)
[CHIPSEC] Platform: Intel Atom Processor C3000 Product Family
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 1980
[CHIPSEC]      RID: 00
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 19DC
[CHIPSEC]      RID: 11
```

Failed module
```
[x][ =======================================================================
[x][ Module: BIOS Interface Lock (including Top Swap Mode)
[x][ =======================================================================
[*] BiosInterfaceLockDown (BILD) control = 0
[*] BIOS Top Swap mode is disabled (TSS = 0)
[*] RTC TopSwap control (TS) = 0
[-] FAILED: BIOS Interface is not locked (including Top Swap Mode)
```
I enabled with_print=1 at chipsec/modules/common/bios_ts.py 
bild = self.cs.get_control( 'BiosInterfaceLockDown' ,with_print=1)

```
[*] running module: chipsec.modules.common.bios_ts
[x][ =======================================================================
[x][ Module: BIOS Interface Lock (including Top Swap Mode)
[x][ =======================================================================
[*] GCS = 0x00000000 << General Control and Status (RCBA + 0x3410)
    [00] BILD             = 0 << BIOS Interface Lock Down 
    [10] BBS              = 0 << Boot BIOS Straps 
[*] BiosInterfaceLockDown (BILD) control = 0
[*] BIOS Top Swap mode is disabled (TSS = 0)
[*] RTC TopSwap control (TS) = 0
[-] FAILED: BIOS Interface is not locked (including Top Swap Mode)
```
But I have BILD = 1 in other module

```
[*] running module: chipsec.modules.common.bios_wp
[x][ =======================================================================
[x][ Module: BIOS Region Write Protection
[x][ =======================================================================
[*] BC = 0xAA << BIOS Control (b:d.f 00:31.5 + 0xDC)
    [00] BIOSWE           = 0 << Write Protect Disable 
    [01] BLE              = 1 << BIOS Lock Enable 
    [02] SRC              = 2 << SPI Read Configuration 
    [04] TSS              = 0 << Top Swap Status 
    [05] SMM_BWP          = 1 << Enable InSMM.STS 
    [07] BILD             = 1 << BIOS Interface Lock-Down 
[+] BIOS region write protection is enabled (writes restricted to SMM)
```

`GCS = 0x00000000 << General Control and Status (RCBA + 0x3410)`
is in common.xml, but I cannot find it in Denverton datasheet. 

Is that a right register for Denverton CPU to get?

NicholasArmour, 2020-01-06:
You are right, BILD is not located in a register defined at (RCBA + 0x3410), we did not have the BiosInterfaceLockDown control defined in dnv.xml, so it defaulted to the one in common.xml, which is incorrect.
[This PR](https://github.com/chipsec/chipsec/pull/812) should fix the issue. I was able to find the register definitions in https://www.intel.com/content/www/us/en/products/docs/processors/atom/c-series/c3000-family-datasheet.html

hans-cameo, 2020-01-15:
> You are right, BILD is not located in a register defined at (RCBA + 0x3410), we did not have the BiosInterfaceLockDown control defined in dnv.xml, so it defaulted to the one in common.xml, which is [...]

This PR fix the issue, thanks.

closed 2020-01-16 by NicholasArmour