chipsec/chipsec #816: System hang+restart when enumerating PCI devices on MacBook Pro 16,1

martingalvan, 2020-01-08:
We've been testing Chipsec on a 2019 MacBook Pro 16,1, and we're seeing that the entire system (including UI) hangs, then restarts for most Chipsec commands. After analyzing this, we've narrowed it down to the PCI enumeration performed in the `enumerate_devices` method at chipsec/hal/pci.py. More precisely, the system hangs when reading the config space of all devices after (and including) BDF 0xFF:3:0.

We tested on some other systems, and saw that the issue only happens on macOS systems that have both Intel processor-integrated graphics and a discrete graphics chip (AMD Radeon on all tested systems.) It only happens when such systems are actively using the Intel processor-integrated graphics. It does NOT happen when using the discrete graphics chip, or on systems that only have integrated graphics.

martingalvan, 2020-01-08:
So we ran some more tests, and arrived to the conclusion that the device and function numbers don't really matter here. The hang seems to be triggered the 25th time we try to read anything from bus 0xFF, even if it's the same B:D:F. As a test, we tried reading B:D:F 25 times; the system hung on the 25th read.

This doesn't seem to be related to timing, either. At one point we saw that the issue was triggered on the 21th read; after a while we realized that the 4 missing reads were from a previous test which didn't hang the system.

skochinsky, 2020-01-08:
Is the system equipped with the T2 chip?

martingalvan, 2020-01-08:
Yes, all the systems have the T2 chip. However, we also tested this on other MacBooks which also have the T2 chip, but not the discrete+integrated GPUs, and couldn't reproduce the issue. We also tested on MacBooks without the T2 chip, but with the discrete+integrated GPUs, and couldn't reproduce the issue either.

The CPU VID/DIDs of the failing systems are 0x8086/0x3E20 and 0x8086/0x3EC4. The PCH VID/DIDs are 0x8086/0xA313 for both systems.

martingalvan, 2020-01-09:
I've tried an lspci build for macOS and saw that the crash doesn't happen when invoking it with -H1. Looking at the code, they seem to do a [recursive scan](https://wiki.osdev.org/PCI#Recursive_Scan), which looks more efficient and doesn't end up hitting bus 0xFF. However, I don't know how reliable is this since we may very well have some bridge pointing to bus 0xFF which will end up breaking things again.

Sae86, 2022-06-14:
Closing due to Mac never being fully supported and since Apple has dropped Intel Silicon, work on it has been deprioritized.

closed 2022-06-14 by Sae86