chipsec/chipsec #821: win32helper: unnecessary check for STATUS_PRIVILEGED_INSTRUCTION?

assafcarlsbad, 2020-01-12:
After issuing an unsuccessful IOCTL to the driver using `win32file.DeviceIoControl`, the helper attempts to compare the error status to `STATUS_PRIVILEGED_INSTRUCTION`. However, nowhere in its IOCTL dispatch routine the driver sets or returns this particular error code. Any chance this check is not relevant anymore?

BrentHoltsclaw, 2020-01-15:
@johnloucaides @ErikBjorge can one of you help answer this question?

johnloucaides, 2021-02-28:
Unless I'm mistaken, it's possible for one of the multiple calls to `GetExceptionCode()` to return STATUS_PRIVILEGED_INSTRUCTION according to docs here: https://docs.microsoft.com/en-us/windows/win32/debug/getexceptioncode

I think this used to come up with some of our fuzzing, actually.

closed 2021-03-07 by assafcarlsbad