chipsec/chipsec #825: Request to add support for AMD Rome CPUs

NewtonLiuNVIDIA, 2020-01-14:
Chipsec uses a register map based on the device's PCIe VID:DID, but doesn't yet have support for the AMD Rome CPUs.  Please help add support for the AMD Rome.

BrentHoltsclaw, 2020-01-23:
There is currently a project to help expand architecture support within CHIPSEC.  Once that work is finished, it should make adding additional support.  Once that is finished, we can support others with guidance.

yaperez-anssi, 2020-07-10:
Hi, I'm currently looking at an AMD platform (Renoir). In order to check the non-CPU related modules (like UEFI variables stuff etc.) I tried to build a really simple AMD configuration with just the DID and ReqPch=False.

It seems that all standard modules get run, even when they don't make sense. For example spi_fdopss has a guard checking if `HSFS` register is defined, but it's stil run in my case while I don't have an HSFS register defined in the configuration. Could it be that default configuration still include some registers and that's the point of the “expand architecture support” project mentioned above?

BrentHoltsclaw, 2020-07-10:
any "common" files still load as a part of all configurations.  I'll see if I can correct the behavior in some upcoming patches.  It may cause some module errors as they are support by family opposed to register until farther processor granularity is added.

kerneis-anssi, 2021-07-26:
Hi, is there any update on this issue? I'd like to try and add some AMD support to chipsec (eg. dumping SPI or checking ROM protection registers) and wonder if you would consider contributions or have any guidance before I start. Thanks!

kerneis-anssi, 2021-07-26:
I just noticed https://github.com/chipsec/chipsec/projects/1 and https://github.com/chipsec/chipsec/issues/461 which may be relevant but have seen no update in the last 18 months.

frinzell, 2021-07-26:
The majority of the needed framework changes are in place to add this support.  If you were to implement this support, we could help address any framework deltas.  Contact us at chipsec@intel.com

kerneis-anssi, 2021-08-11:
For the record, I've reached out and I'm starting some work to support AMD Renoir (technically the CPU I have at hand is Cezanne, but it looks like there is no significant difference between the two). Feel free to rename this issue to remove "Rome" from it and assign it to me. #461 can also probably be closed as a duplicate.

headkaze, 2021-08-31:
> For the record, I've reached out and I'm starting some work to support AMD Renoir (technically the CPU I have at hand is Cezanne, but it looks like there is no significant difference between the [...]

I also have a Cezanne CPU. I just tried your 'amd' branch and I get the following error:
```
$ sudo python3 chipsec_util.py -i mmio list

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.7.0
[CHIPSEC] OS      : Linux 5.14.0-custom #1 SMP Tue Aug 31 00:25:42 EDT 2021 x86_64
[CHIPSEC] Python  : 3.9.5 (64-bit)

****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
ERROR: Unknown Platform: VID = 0x1022, DID = 0x1630, RID = 0x00
WARNING: *******************************************************************
WARNING: * Unknown platform!
WARNING: * Platform dependent functionality will likely be incorrect
WARNING: * Error Message: "Unknown Platform: VID = 0x1022, DID = 0x1630, RID = 0x00"
WARNING: *******************************************************************
[CHIPSEC] Helper  : LinuxHelper (/home/***/Programming/chipsec/chipsec/helper/linux/chipsec.ko)
[CHIPSEC] Platform: UnknownPlatform
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
[CHIPSEC] Executing command 'mmio' with args ['list']
```

kerneis-anssi, 2021-08-31:
For the record, this branch currently doesn't have any AMD-specific modules, it is a work-in-progress. I'll submit a PR once it's ready for some more serious testing. You can cherry-pick the following commit if you want to experiment on your own: https://github.com/chipsec/chipsec/commit/48daf67eab6ab14461c64dcd17f7857978c42c56 (note that this XML config will soon be completely rewritten, at which point I'll push to the `amd` branch instead).

headkaze, 2021-08-31:
Thanks @kerneis-anssi I can successfully get a dump of SPIBAR. 
```
$ sudo python3 chipsec_util.py mmio dump SPIBAR

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.7.0
[CHIPSEC] OS      : Linux 5.14.0-custom #1 SMP Tue Aug 31 00:25:42 EDT 2021 x86_64
[CHIPSEC] Python  : 3.9.5 (64-bit)

****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[!]       Unknown PCH: VID = 0x1022, DID = 0x790E, RID = 0xFF; Using Default.
[!]            Results from this system may be incorrect.
[CHIPSEC] Helper  : LinuxHelper (/home/***/Programming/chipsec/chipsec/helper/linux/chipsec.ko)
[CHIPSEC] Platform: Renoir Root Complex
[CHIPSEC]      VID: 1022
[CHIPSEC]      DID: 1630
[CHIPSEC]      RID: 00
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
[CHIPSEC] Executing command 'mmio' with args ['dump', 'SPIBAR']

[CHIPSEC] Dumping SPIBAR MMIO space..
[mmio] MMIO register range [0x00000000FEC10000:0x00000000FEC10000+000000FF]:
+00000000: 4F0C2105
+00000004: 00000000
+00000008: 00000600
+0000000C: 02000000
+00000010: 04042006
+00000014: 059F0406
+00000018: 020A0B03
+0000001C: 0206B8FF
+00000020: 10330713
+00000024: 20202008
+00000028: 0E06140C
+0000002C: 800054C0
+00000030: 460814CC
+00000034: 00000003
+00000038: FCFCFCFC
+0000003C: 000088FC
+00000040: EBBB6B3B
+00000044: 00000500
+00000048: 02000001
+0000004C: 00030002
+00000050: 0C131200
+00000054: ECBC6C3C
+00000058: 00004608
+0000005C: 00000000
+00000060: 00000000
+00000064: 000000FD
+00000068: 00000000
+0000006C: 00000000
+00000070: 00000000
+00000074: 00000000
+00000078: 00000000
+0000007C: 00000000
+00000080: 05000000
+00000084: DA001123
+00000088: 300665B1
+0000008C: C95F17BB
+00000090: 870ED138
+00000094: 0324F3C3
+00000098: F01DFA28
+0000009C: 95E8B95B
+000000A0: D75105C2
+000000A4: 82FEDDA5
+000000A8: 5CE01B8E
+000000AC: 8CF1F388
+000000B0: 07614F3A
+000000B4: 9D89E91B
+000000B8: 84145D79
+000000BC: 81DC60B7
+000000C0: FF765F24
+000000C4: 00000000
+000000C8: 00000000
+000000CC: 00000000
+000000D0: 00000000
+000000D4: 00000000
+000000D8: 00000000
+000000DC: 00000000
+000000E0: 00000000
+000000E4: 00000000
+000000E8: 00000000
+000000EC: 00000000
+000000F0: 00000000
+000000F4: 00000000
+000000F8: 00000000
[CHIPSEC] (mmio) time elapsed 0.002
```

headkaze, 2021-09-14:
@kerneis-anssi I added SPI control registers in my [pull request](https://github.com/kerneis-anssi/chipsec/pull/2). Not sure if you missed it?

kerneis-anssi, 2021-09-14:
Completely missed it, thanks for the help! Will have a look tomorrow. Just so I understand better who's who, are you working for Intel or on chipsec?

Les données à caractère personnel recueillies et traitées dans le cadre de cet échange, le sont à seule fin d’exécution d’une relation professionnelle et s’opèrent dans cette seule finalité et pour la durée nécessaire à cette relation. Si vous souhaitez faire usage de vos droits de consultation, de rectification et de suppression de vos données, veuillez contacter ***@***.*** Si vous avez reçu ce message par erreur, nous vous remercions d’en informer l’expéditeur et de détruire le message. The personal data collected and processed during this exchange aims solely at completing a business relationship and is limited to the necessary duration of that relationship. If you wish to use your rights of consultation, rectification and deletion of your data, please contact: ***@***.*** If you have received this message in error, we thank you for informing the sender and destroying the message.

headkaze, 2021-09-14:
> Completely missed it, thanks for the help! Will have a look tomorrow. Just so I understand better who's who, are you working for Intel or on chipsec?

I'm only interested in helping add AMD support to chipsec for personal use (BIOS hacking). I got the registers from the [AMD SB600 
Register Reference Manual](https://www.amd.com/system/files/TechDocs/46155_sb600_rrg_pub_3.03.pdf).

headkaze, 2021-09-19:
@kerneis-anssi I've updated [renoir.xml](https://github.com/headkaze/chipsec/blob/amd/chipsec/cfg/1022/renoir.xml) to include SPI and ESPI registers.

kerneis-anssi, 2021-11-23:
@headkaze: thanks for reporting the issue with the Renoir XML file (missing desc fields): https://github.com/chipsec/chipsec/pull/1259#issuecomment-976027289
Can you please test the latest commit I pushed to my `amd` branch?

headkaze, 2021-11-23:
It's working for me
```
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.7.3
[CHIPSEC] OS      : Linux 5.15.0-acso #1 SMP Mon Nov 1 17:43:04 EDT 2021 x86_64
[CHIPSEC] Python  : 3.9.7 (64-bit)

****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[!]       Unknown PCH: VID = 0x1022, DID = 0x790E, RID = 0xFF; Using Default.
[!]            Results from this system may be incorrect.
[CHIPSEC] Helper  : LinuxHelper (/home/bbaker/Programming/temp/chipsec/chipsec/helper/linux/chipsec.ko)
[CHIPSEC] Platform: Renoir Root Complex
[CHIPSEC]      VID: 1022
[CHIPSEC]      DID: 1630
[CHIPSEC]      RID: 00
[CHIPSEC] PCH     : Default PCH
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
[CHIPSEC] Executing command 'mmio' with args ['list']

--------------------------------------------------------------------------------------
 MMIO Range   | BUS | BAR Register   | Base             | Size     | En? | Description
--------------------------------------------------------------------------------------
 IOMMUBAR     |  00 | IOMMU_CAP_BASE_LO | 00000000FD300000 | 00100000 | 1   | IOMMU Base Address
 ESPI         |  00 | SPIBaseAddr + 0x10000 | 00000000FEC20000 | 00000100 | 1   | eSPI Controller Register Range
 SDHC         |  00 | SD_PCI_BAR     | FFFFFFFFFFFFFF00 | 00000100 | 1   | SD Host Controller Base Address
 SPI          |  00 | SPIBaseAddr    | 00000000FEC10000 | 00000100 | 1   | SPI Controller Register Range
 IOMMU_MMIO_DEVTBL_BASE |  00 | IOMMU_MMIO_DEVTBL_BASE_0 | 0000000001400000 | 00001000 | 1   |
```

AkechiShiro, 2022-04-09:
Hey I would like to port my Picasso AMD Ryzen 7 3700U CPU's to work with chipsec is there any guide on how to do this, I didn't find much documentation on how to add support for a CPU ? Or should I try to understand what's been done by @kerneis-anssi so far in the commits made on his fork ?

kerneis-anssi, 2022-04-11:
@AkechiShiro You should just need to create a distinct xml file describing the registers you care about for the platform in https://github.com/chipsec/chipsec/tree/main/chipsec/cfg/1022. The current renoir.xml is not the best example, it should identify the CPU family based on cpuid instead of the PCI device id. I'll try and send a PR later this week to fix that (but it's a busy week…).

What are you interested in exactly? I'd be happy to help or collaborate. Don't hesitate to reach out directly to me (gabriel.kerneis@ssi.gouv.fr).

AkechiShiro, 2022-04-11:
@kerneis-anssi I confirm that for the Picasso CPUID, it is `0x0810F81` hence you were right.
More info on my exact platform : 
```
[CHIPSEC] Platform: Actual values: VID = 0x1022, DID = 0x15D0, RID = 0x00, CPUID = 0x0810F81
```

I'd like to fully implement all the applicable modules (first the common ones, and then think about what specific modules there would be) for an AMD Ryzen 7 PRO 3700U, in order to better assess the hardware security of my device, testing with `-p RENOIR` shows that 7 modules failed with an exception, 12 fails just to run, 5 passed, 9 were not applicable, 1 is not applicable.

I'm wondering how is it decided that a module is not applicable, based on what information could one figure out that the SMM memory is not accessible by the kernel ? What is the difference with an Intel CPU, as I think some Intel platforms can access SMM memory via the kernel, making me question why under an AMD CPU that wouldn't be possible. 
Could this be due to the fact that the equivalent of the Intel Management Engine under AMD; AMD Platform Security Processor (PSP) works differently than Intel ME ?

Thanks for your help on getting started, I'll be sure to reach out to you directly with more details, in order to improve the support for my CPU as well as if I do run in any specific issues, maybe I can help to contribute as well to the Renoir platform.

Finally, one last thing, I think it would be something nice to have, is to add a platform specific chipsec module to detect [TSME](https://en.wikichip.org/wiki/x86/sme#Transparent_SME) and maybe [SEV](https://en.wikichip.org/wiki/x86/sme#Secure_Encrypted_Virtualization) for AMD CPUs, there is this code, that could [test for TSME support on AMD platforms](https://github.com/AMDESE/mem-encryption-tests).

kerneis-anssi, 2022-04-12:
The currently committed changes only supports introspecting registers through `chipsec_util reg read` and other chipsec_util commands. The security check modules of `chipsec_main` have not been ported and the results are meaningless.

I also want to add support to security check modules, so let's work on that together :-) To get started, pick this PR: https://github.com/chipsec/chipsec/pull/1396. It will disable irrelevant modules (basically, only UEFI variable checks are meaningful)

The next steps would consist in:
- deciding what AMD security feature you want to check
- create a bug to track the implementation of this feature and add it to the following project: https://github.com/chipsec/chipsec/projects/2. We can then discuss each feature on a specific bug.
- figure out what registers to read, or how to check it (speaking to the PSP directly? walking some structure in memory?)
- adding a module in chipsec/modules/1022 that performs those checks; you can copy a module from chipsec/modules/8086 to get the general structure, but the implementation will necessarily differ (what registers to read, etc.)
- if you need to perform some kind of operation that is not already supported by chipsec (eg. talking to the PSP), add a module in chipsec/hal to implement primitives allowing you to do that (eg. chipsec/hal/psp.py) and use it in your security-check module.

My goal as a first step and example was to implement SPI lock checks: https://github.com/chipsec/chipsec/issues/1319
This doesn't prevent you from working on other features in parallel.

Useful resources: fwupd is adding similar checks, eg. https://github.com/fwupd/fwupd/pull/4471. They are based on a Linux driver ([drivers/crypto/ccp](https://github.com/torvalds/linux/tree/master/drivers/crypto/ccp)) that talks to the PSP. We don't want to use that kernel module in chipsec (because we need to be portable), but we want to port its core functionality to chipsec/hal modules.

kerneis-anssi, 2024-10-30:
While this doesn't address Rome specifically, anybody interested in improved AMD support should have a look at https://github.com/chipsec/chipsec/pull/2343