chipsec/chipsec #831: tools.uefi.blacklist: can not allocate memory

fpusersuggest, 2020-01-20:
Hi, I suspect to have uefi rootkit and a guy has gave me the advice of to check it with chipsec and the module tools.uefi.blacklist, but it gave me an error:
                                                              
```
[*] dumping FW image from ROM to fw.bin: 0x00200000 bytes at [0x007FFFFF:0x00600000]
[*] this may take a few minutes (instead, use 'chipsec_util spi dump')...
[*] searching for EFI binaries that match criteria from 'blacklist.json':
    HT_UEFI_Rootkit  - HackingTeam UEFI Rootkit (https://www.mcafee.com/enterprise/en-us/threat-center/advanced-threat-research/uefi-rootkit.html)                                                                                
    ThinkPwn         - ThinkPwn: SystemSmmRuntimeRt SMM vulnerability (http://blog.cr4.sh/2016/06/exploring-and-exploiting-lenovo.html)                                                                                           
    LoJax            - LoJax: First UEFI rootkit found in the wild (https://www.welivesecurity.com/2018/09/27/lojax-first-uefi-rootkit-found-wild-courtesy-sednit-group/)                                                         
Decoding                                                                                                             
Error: Can not allocate memory           
Decoding                                                                  
Error: Can not allocate memory                         
Decoding                                                                                                                                                                
Error: Can not allocate memory

```

Anyway the following is the complete LOG:

```
# ./chipsec_main.py  -m tools.uefi.blacklist
################################################################                                   
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##                    
##                                                            ##         
################################################################         
[CHIPSEC] Version 1.4.4                                                                                                                                                                                                           
[CHIPSEC] Arguments: -m tools.uefi.blacklist                                                                                                                                                                                      
****** Chipsec Linux Kernel module is licensed under GPL 2.0                                                                                                                                                                      
[CHIPSEC] API mode: using CHIPSEC kernel module API                         
[CHIPSEC] OS      : Linux 5.0.0-31-generic #33-Ubuntu SMP Mon Sep 30 18:51:59 UTC 2019 x86_64
[CHIPSEC] Python  : 2.7.16 (64-bit)
[CHIPSEC] Helper  : LinuxHelper (/home/lws/chipsec-master/chipsec/helper/linux/chipsec.ko)
[CHIPSEC] Platform: Braswell/Cherry Trail SoC                             
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 2280                               
[CHIPSEC]      RID: 35                                                                   
[CHIPSEC] PCH     : Default PCH                                                                       
[CHIPSEC]      VID: 8086                                                            
[CHIPSEC]      DID: 229C      
[CHIPSEC]      RID: 35
        
[+] loaded chipsec.modules.tools.uefi.blacklist
[*] running loaded modules .. 
                                                                        
[*] running module: chipsec.modules.tools.uefi.blacklist                                
[x][ =======================================================================
[x][ Module: Check for black-listed EFI binaries in UEFI firmware                    
[x][ =======================================================================
                                                                                 
Usage:                                                                      
                                                                           
    chipsec_main.py -i -m tools.uefi.blacklist [-a <fw_image>,<blacklist>]      
                                                                                                   
      fw_image  : Full file path to UEFI firmware image
                  If not specified, the module will dump firmware image directly from ROM                                               
      blacklist : JSON file with configuration of black-listed EFI binaries (default = blacklist.json)
                  Config file should be located in the same directory as this module
                                       
Examples:                           
                                    
    chipsec_main.py -m tools.uefi.blacklist
                                    
      Dumps UEFI firmware image from flash memory device, decodes it and
      checks for black-listed EFI modules defined in the default config 'blacklist.json'
    chipsec_main.py -i --no_driver -m tools.uefi.blacklist -a uefi.rom,blacklist.json

      Decodes 'uefi.rom' binary with UEFI firmware image and
      checks for black-listed EFI modules defined in 'blacklist.json' config
[... 71 lines cut ...]
Decoding

Error: Can not allocate memory

^CERROR: Exception occurred during chipsec.modules.tools.uefi.blacklist.run(): ''

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            113.087
[CHIPSEC] Modules total           1
[CHIPSEC] Modules failed to run   1:
ERROR: chipsec.modules.tools.uefi.blacklist
[CHIPSEC] Modules passed          0:
[CHIPSEC] Modules information     0:
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   0:
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  0:
[CHIPSEC] Modules with exceptions 1:
ERROR: chipsec.modules.tools.uefi.blacklist
[CHIPSEC] *****************************************************************
```

Thank you very much.

IF ANYWAY WANT THE UEFI IMAGE BACKUP I CAN SEND IT.

NicholasArmour, 2020-01-21:
This looks like a problem with our `uefi decode`. 

Can you send us the UEFI image?

0xfede7c8, 2020-01-30:
Almost every image triggers these messages in the UEFI parser.

fpusersuggest, 2020-01-31:
@NicholasArmour did u get mail email with the attachment ? 
(please don't publish my email address)

NicholasArmour, 2020-02-03:
Yes I received your email, we are looking into it.

fpusersuggest, 2020-02-07:
Hi, a question, how chipsec prevents an infected kernel to give it a fake uefi backup ?
or is there a software way to have a trusted backup ?

BrentHoltsclaw, 2020-02-07:
CHIPSEC makes no claims to prevent/protect data from an infected kernel.  If the kernel is suspected to be infected, alternate methods to collect data would be live image or physically reading from the spi chip.

BrentHoltsclaw, 2020-02-07:
I am unable to reproduce the errors with python2.  With python3 I am seeing some bytestring errors, but those seem to be unrelated to the issue reported.

fpusersuggest, 2020-02-08:
If the kernel is infected by a uefi rootkit is not the same with a live image ? I mean if the uefi rootkit can infect (theoretically) a kernel of an installed system can infect a kernel of live image, isn't it ? 
For the bug I have to reboot on linux to find as much as possible info.

fpusersuggest, 2020-02-09:
Hi, I have python2 but I have discovered that if I leave the software run completes the analysis, I mean it take about 30 seconds with hundreds of *Error: Can not allocate memory* but it completes the analysis.
Anyway I have python2 on ubuntu 19.04 and kernel 5.0.0.0-31. And my laptop has only 2 GB of ram.

BrentHoltsclaw, 2020-02-12:
> I mean if the uefi rootkit can infect (theoretically) a kernel of an installed system can infect a kernel of live image, isn't it ?

Yes, you could try reading from the spi directly

BrentHoltsclaw, 2020-02-12:
> _Error: Can not allocate memory_

This error is a result of calling the compression tools with a piece of the firmware image as we are not sure which compression algorithm it is using.  We currently do not have a good way to suppress the output from the calls.