chipsec/chipsec #846: Denverton issue : ME is in Manufacturing Mode

jf6316, 2020-02-05:
We use Intel Denverton C3708.
Get failed at a module “ME is in Manufacturing Mode”
Does the register CHIPSEC try to get exists in Denverton CPU?
In common.xml I find define 
`<register name="HFS" type="pcicfg" bus="0" dev="0x16" fun="0" offset="0x40" size="4" desc="ME Host Firmware Status">`
But in Intel Denverton EDS Doc #558579 **I can't find B0:D22:F0 this item.** 
Are this defines correct for Denverton CPU?

Our hardware has been set "0" to the IERR_N pin for Flash descriptor security locked.
[image]

Here is the log:
```
[*] running module: chipsec.modules.common.me_mfg_mode
FSOpen: Open '\chipsec' Success
FSOpen: Open '\chipsec' Success
FSOpen: Open 'chipsec\modules\common\me_mfg_mode.py' Success
FSOpen: Open '\chipsec' Success
FSOpen: Open '\chipsec' Success
FSOpen: Open '\chipsec' Success
FSOpen: Open '\chipsec' Success
FSOpen: Created new directory entry 'me_mfg_mode.pyc'
FSOpen: Open 'chipsec\modules\common\me_mfg_mode.pyc' Success
[x][ =======================================================================
[x][ Module: ME Manufacturing Mode
[x][ =======================================================================
[-] FAILED: ME is in Manufacturing Mode
```

BrentHoltsclaw, 2020-02-05:
According to  datasheet 337018-002.  52.2.16 shows 
```
Type: PCI Configuration Register
(Size: 32 bits)
Offset: [B:0, D:24, F:0] + 40h
```
Will open a PR with the change

BrentHoltsclaw, 2020-02-05:
Can you let me know if 847 fixes the issue

jf6316, 2020-02-06:
After 847 fixes the issue has been fixed.
Here is log:
```
[*] running module: chipsec.modules.common.me_mfg_mode
FSOpen: Open '\chipsec' Success
FSOpen: Open '\chipsec' Success
FSOpen: Open 'chipsec\modules\common\me_mfg_mode.py' Success
FSOpen: Open '\chipsec' Success
FSOpen: Open 'chipsec\modules\common\me_mfg_mode.pyc' Success
[x][ =======================================================================
[x][ Module: ME Manufacturing Mode
[x][ =======================================================================
[+] PASSED: ME is not in Manufacturing Mode
```

BrentHoltsclaw, 2020-02-12:
Issue appears to be fixed and the code is merged in.

closed 2020-02-12 by BrentHoltsclaw