chipsec/chipsec #870: Possible to extract ROM sections from Lenovo BIOS updates (to apply me_cleaner on FL2 extraction?)

tlaurion, 2020-03-13:
As specified [here](https://github.com/osresearch/heads/issues/307#issuecomment-598913489):

I would have expected to be able to extract ME region ecognized by me_cleaner:
`wget https://download.lenovo.com/pccbbs/mobiles/g2uj31us.exe && innoextract ~/g2uj31us.exe && echo -e "\n\n EXTRACTING FL1\n\n\n" && echo -e "\n\n EXTRACTING FL1\n\n\n" && python chipsec_util.py -n uefi decode ./app/G2ETB5WW/\$01D3000.FL1 && echo -e "\n\n\n EXTRACTING FL2 \n\n\n" && python chipsec_util.py -n uefi decode ./app/G2ETB5WW/\$01D3000.FL2 && echo -e "\n\n\n Testing me_cleaner against all extracted files.... output given only on successful files... \n\n\n" && find ./app/G2ETB5WW/\$01D3000.FL*.dir/ -type f | while read filename; do python ~/me_cleaner/me_cleaner.py $filename &>/dev/null; if [ $? -eq 0 ]; then echo $filename; fi; done;`

But nope. me_cleaner doesn't pickup any me segment to play with, no output.

Is there an unknown tool out there able to accomplish this job?

tlaurion, 2020-03-13:
@tweksteen @rountree @hughsie @WilliamLeara How to accomplish [this](https://github.com/osresearch/heads/issues/307#issuecomment-598508208)?

rountree, 2020-03-13:
@tlaurion, sorry, that's several zip codes away from my competence.  No idea.

n4ru, 2020-03-14:
Binwalk to find the offset, then dd out the length of the uefi modules.

I am not sure if ME is shipped with the FL2 file if you’re referring to bios updates (FL2 is EC region, maybe more but have not checked).

BIOS is in FL1.

However, the ME update is a separate program. Perhaps you can find it there?

skochinsky, 2020-03-14:
Yes, FL2 in this update is only ~190K, way too small to contain the ME firmware. FL1 seems to contain only the BIOS update capsule and some padding.

tlaurion, 2020-03-15:
@n4ru : We can't extract the whole 8mb SPI flash image somehow?

skochinsky, 2020-03-19:
You can't extract something which is not there no matter which tool you use. There is no real need for Lenovo to distribute the full image so they ship the relevant parts separately (one update for EC+ BIOS, another for ME). You'll have to use a programmer if you want the full image.

n4ru, 2020-03-19:
As @skochinsky said, it's likely Lenovo only ships the ME binary itself, and the full UEFI for the EC/BIOS since that is updated entirely in one go.

IFD and GbE at least will still have to be extracted.

step21, 2022-01-20:
At least on my t450s - FL1 is smaller, so it is not the full UEFI either. (a full dump is 16 MB, whereas FL1 in the latest update is 6,9 MB) In some other project/thread it was suggested to copy the parts over from the update by hand... though I only need to clear nvram in principle.

tlaurion, 2022-01-22:
Sorry I forgot this thread @n4ru @rountree @skochinsky @step21 

xx20 extraction script https://github.com/osresearch/heads/blob/master/blobs/xx20/download_parse_me.sh

xx30 extraction script: https://github.com/osresearch/heads/blob/master/blobs/xx30/download_clean_me.sh

GBE generation instructions inside of README.

IFD still needs to ne extracted, but not considered problematic for redistribution so provided in tree over there.

The result is the creation of fully externally flashable ROMs for xx20 and xx30 boards currently supported under Heads through maximized builds.

Maximized builds are taking neutered ME space (4.5Mb) and assigning it to IFD BIOS space region. On xx30, the IFD provided being unlocked permits Heads internal upgrades of combined 8mb bottom SPI and 4Mb top SPI. But it requires an initial external flash of those splitted ROMs on their corresponding SPI. The result for xx30 is that 11.5Mb is useable for coreboot combined builds, including the payload, as opposed to 7Mb if ME is not neutered and that space is not given back to the BIOS region under IFD.

GBE in tree has Ethernet MAC fixed to DE:AD:C0:FF:EE. So Mac randomization is needed prior of bringing up interface and being used.

xx20: x220, t420, w520.
xx30: x230, t430, w530.

tlaurion, 2022-01-22:
And as for this raised issue, the complete ME blob is not directly available for xx20, where me_cleaner needed to be modified to be able to extract ME directly from Lenovo website.

For xx30, it worked right away.

I have not checked for newer Lenovo, since I'm not really interested personally in platforms needing blobs in the boot path (FSP, graphic init, MRC blob and ME not being neeuterable after Broadwell.)

Thanks for having updated this thread. Hopefully this will be useful for others.

As a pointer, its was neither FL1 or FL2 that contained ME. But another update exe which could be extracted and manipulated directly. Check the scripts in above post.

On my side this can be closed. The solution didn't have anything related to chipsec at all.

Sae86, 2022-06-06:
Closing this issue as per above comment.

closed 2022-06-06 by Sae86