chipsec/chipsec #877: PCIEXBAR usage?

martingalvan, 2020-03-27:
Hi all, I'm writing because I saw the following code in mmio.py:

```
    def get_MMCFG_base_address(self):
        (bar_base,bar_size)  = self.get_MMIO_BAR_base_address('MMCFG')
        # @TODO: temporary w/a
        #if (Cfg.PCI_PCIEXBAR_REG_LENGTH_256MB == (bar_base & Cfg.PCI_PCIEXBAR_REG_LENGTH_MASK) >> 1):
        #    bar_base &= ~(Cfg.PCI_PCIEXBAR_REG_ADMSK128|Cfg.PCI_PCIEXBAR_REG_ADMSK64)
        #elif (Cfg.PCI_PCIEXBAR_REG_LENGTH_128MB == (bar_base & Cfg.PCI_PCIEXBAR_REG_LENGTH_MASK) >> 1):
        #    bar_base &= ~Cfg.PCI_PCIEXBAR_REG_ADMSK64
        ##elif (Cfg.PCI_PCIEXBAR_REG_LENGTH_64MB == (bar_base & Cfg.PCI_PCIEXBAR_REG_LENGTH_MASK) >> 1):
        ##   pass
        if logger().HAL: logger().log( '[mmcfg] Memory Mapped CFG Base: 0x{:016X}'.format(bar_base) )
        return bar_base, bar_size
```

It looks like what Chipsec calls `MMCFG` is actually the PCIEXBAR register, which is defined as follows in common.xml:

`    <bar name="MMCFG"    bus="0" dev="0"    fun="0" reg="0x60" width="8" mask="0x7FFFFFF000" size="0x1000" enable_bit="0" desc="PCI Express Register Range"/>`

I don't understand the reason for commenting out the usage of the `ADMSK*` fields in favor of the BAR `mask`, nor why the mask takes bits 12-onwards, i.e. right in the middle of the Reserved bits for all the CPUs I've seen. Can anyone comment on this?

Thanks!

BrentHoltsclaw, 2020-03-28:
I don't have the project history to tell you why the code was commented out.

Sae86, 2022-06-14:
Closing as fixed per PR #879