chipsec/chipsec #905: Regression in 1.5.0: KeyError 32092

yaperez-anssi, 2020-05-04:
Hi,
I noticed a regression in the recently released 1.5.0. On two machines (Lenovo ThinkPad X280 and HP Probook 450 G7) when I try to run chipsec_main it fails with:

  KeyError: 32902

The traceback points to chipsec/chipset.py:224 with vid being 32902 (so 0x8086 which is consistent with the chipsets). I'll try to run a bisect to identify the issue but I wanted to report early in case someone else has a clue.

Also should I run python3 setup.py install at each bisect step or is there a faster way to test?

yaperez-anssi, 2020-05-04:
The bisect was pretty short, here is the result log:

```
# bad: [2c6528a38faa54bbf9e7e4483836b75449606639] Update to verison 1.5.0
# good: [604c3fc8b91c8e71705a39bfe93ee4a475d4424a] Update to version 1.4.9
git bisect start '1.5.0' '1.4.9'
# bad: [6d491162bf8b7ed37217c94bb3e4c5d00a8e70df] linux helper HAS_EFI bugfix
git bisect bad 6d491162bf8b7ed37217c94bb3e4c5d00a8e70df
# good: [12bb725351e45a631eb192a556ed440c9cc0be0a] Update copyright
git bisect good 12bb725351e45a631eb192a556ed440c9cc0be0a
# bad: [55a3f7e700df34109b477493c6dea0a3ca05bb79] Changes after rebase with 1.4.8
git bisect bad 55a3f7e700df34109b477493c6dea0a3ca05bb79
# skip: [d88dc907f3ef1abd176b61c36dd3f998c78ff384] Modify chipset.py to load configuration dictionary from xml
git bisect skip d88dc907f3ef1abd176b61c36dd3f998c78ff384
# good: [194c9ec70bc18d0b4317e325757b3d582fd1bd50] Reorganize xml
git bisect good 194c9ec70bc18d0b4317e325757b3d582fd1bd50
# bad: [514c23d84d36a9bfe9f1d650db1a33489c527782] Modify files based upon chipset.py changes
git bisect bad 514c23d84d36a9bfe9f1d650db1a33489c527782
# only skipped commits left to test
# possible first bad commit: [514c23d84d36a9bfe9f1d650db1a33489c527782] Modify files based upon chipset.py changes
# possible first bad commit: [d88dc907f3ef1abd176b61c36dd3f998c78ff384] Modify chipset.py to load configuration dictionary from xml
```

The two pointed commits look indeed relevant. I'll try to investigate a bit more later but any pointer appreciated.

BrentHoltsclaw, 2020-05-04:
Can you elaborate on how you updated and that you have the latest xml configuration files loaded.  The chipset_dictionary should be populated from the individual xml files within the cfg/8086 directory.  Also if you can send a log with the -d flag, that would help let us know if there are any problems within the loading of the xml as it silently fails.

yaperez-anssi, 2020-05-05:
Hi, I'm actually using Chipsec in a custom-built environment for testing machines as part of a tender process. The build process is here: https://github.com/ANSSI-FR/chipsec-check/blob/master/tools/create-chipsec.sh

We basically debootstrap a Debian then clone Chipsec and install it. There's an ISO published some time ago (https://github.com/ANSSI-FR/chipsec-check/releases/download/v1.2/chipsec.iso)  should include master at the time it was built and is working. I tried to rebuild a new version recently (which includes Chipsec 1.5.0) and it exhibits the error above (I'll try to provide the full backtrace later). 

Then I tried to bisect between 1.4.9 (which I confirmed working) and 1.5.0 and go the above log.

I've first tried 1.5.0 on the HP Probook 450 G7 which was unsupported platform on 1.4.9, so it might be relevant.

BrentHoltsclaw, 2020-05-05:
When the `setup.py install` option is used it is not copying over the `chipsec/cfg/8086` directory.

yaperez-anssi, 2020-05-05:
I'm confused then. What is the correct method for installing properly then? The manual points to `setup.py install`, what would be the right thing to do?

Also note that when generating the ISO the `python setup.py install` is only done once, so I'd assume at least some part of the configuration is copied somewhere usable?

BrentHoltsclaw, 2020-05-05:
That is the a correct method for install.  Will work on getting a fix ASAP.  It was missed within the preparation for the last release.  An alternate method used by some is the `setup.py build_ext -i` option.

yaperez-anssi, 2020-05-05:
Ah ok sorry I didn't understand you acknowledged a bug :) Thanks for the tip, I'll try it and report back

BrentHoltsclaw, 2020-05-05:
#907 should solve the issue

yaperez-anssi, 2020-05-05:
I've tried the commit in #907 and I can confirm that I have a `/usr/local/lib/python*/dist-packages/chipsec-1.5.0*/chipsec/cfg` folder which has more XML stuff than before (previously I only had template.xml).

But I still have the same error as previously. Here's the log with the backtrace (ran with -d but it fails early so there's not much more in the log):

```
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.5.0
[CHIPSEC] Arguments: -d

[helper] Linux Helper created
Module /dev/chipsec loaded successfully
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[helper] Linux Helper started/loaded
[CHIPSEC] API mode: using CHIPSEC kernel module API
[-] Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.5.0-py3.7-linux-x86_64.egg/chipsec_main.py", line 469, in main
    self._cs.init( self._platform, self._pch, (not self._no_driver), self._driver_exists, self._to_file, self._from_file )
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.5.0-py3.7-linux-x86_64.egg/chipsec/chipset.py", line 224, in init
    if did in self.chipset_dictionary[vid] and len(self.chipset_dictionary[vid][did]) > 1:
KeyError: 32902

```

npmitche, 2020-05-06:
There was is a new commit to #907 that should now fix it. If you could give it another try and let us know, that would be awesome.

yaperez-anssi, 2020-05-07:
I tried it, as well as running in place (with `python3 setup.py build_ext -i; python3 chipsec_main.py -i`) and both fail with:

```
[*] Ignoring unsupported platform warning and continue execution
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.5.0
[CHIPSEC] Arguments: -i

****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[-] Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.5.0-py3.7-linux-x86_64.egg/chipsec_main.py", line 469, in main
    self._cs.init( self._platform, self._pch, (not self._no_driver), self._driver_exists, self._to_file, self._from_file )
  File "/usr/local/lib/python3.7/dist-packages/chipsec-1.5.0-py3.7-linux-x86_64.egg/chipsec/chipset.py", line 269, in init
    data_dict           = self.pch_dictionary[self.vid][self.pch_did][0]
KeyError: 65535
```
This is still on the HP ProBook 450 G7. I guess 65536 is 0xFFFF but I'm unsure if it's `self.vid` or `self.pch_did` yet.

npmitche, 2020-05-07:
Does both ways have the same path in the trace: `/usr/local/lib/python3.7/dist-packages/chipsec-1.5.0-py3.7-linux-x86_64.egg`? If so, in the downloaded folder, run `sudo python3 ./chipsec_main.py` (Note the "./")

Can you share what's in the `/usr/local/lib/python3.7/dist-packages/chipsec-1.5.0-py3.7-linux-x86_64.egg/chipsec/cfg` folder? Does it match `~/Downloads/Chipsec-1.5.0/chipsec/cfg`?

yaperez-anssi, 2020-05-07:
Good point, I missed that. I retried with `python3 ./chipsec_main.py` but same thing happens (with paths pointing to the current dir, which has the clone (so everything in cfg/ comes from that).

yaperez-anssi, 2020-05-07:
I tried to print-debug, and it seems that the following patch:
```
diff --git a/chipsec/chipset.py b/chipsec/chipset.py
index 57f93c3..3015aee 100644
--- a/chipsec/chipset.py
+++ b/chipsec/chipset.py
@@ -266,13 +266,13 @@ class Chipset:
             self.pch_vid = pch_vid
             self.pch_did = pch_did
             self.pch_rid = pch_rid
-            data_dict           = self.pch_dictionary[self.vid][self.pch_did][0]
+            data_dict           = self.pch_dictionary[self.pch_vid][self.pch_did][0]
             self.pch_code       = data_dict['code']
             self.pch_longname   = data_dict['longname']
             _unknown_pch = False
 
         if _unknown_platform and start_driver:
-            msg = 'Unsupported Platform: VID = 0x{:04X}, DID = 0x{:04X}, RID = 0x{:02X}'.format(self.vid,self.did,self.rid)
+            msg = 'Unsupported Platform: VID = 0x{:04X}, DID = 0x{:04X}, RID = 0x{:02X}'.format(self.pch_vid,self.pch_did,self.pch_rid)
             logger().error( msg )
             raise UnknownChipsetError (msg)
         if not _unknown_platform: # don't intialize config if platform is unknown
```
would fix the issue. I didn't really investigate what is self.pch_vid wrt. self.vid but it seems to progress further (I now get an unsupported platform message for 8086:0284 but it runs with -i).

npmitche, 2020-05-08:
Without the `-i` flag are you seeing the same `KeyError: 65535`?

yaperez-anssi, 2020-05-11:
Since I was a bit confused I tried to print pch_vid,pch_did,pch_rid along with vid,did,rid on a supported (ThinkPad X280) and an unsupported (HP ProBook 450 G7 8086:9b61).

On X280):
```
pch_vid:	 0x8086 pch_did:	 0x9d4e pch_rid:	 0x21
self.vid:	 0x8086 self.did:	 0x5914 self.rid:	 0x8
````
A quick check with lspci:
````
00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers [8086:5914] (rev 08)
00:1f.0 ISA bridge [0601]: Intel Corporation Sunrise Point LPC Controller/eSPI Controller [8086:9d4e] (rev 21)
````
On the HP ProBook 450G7 I have:
````
pch_vid:	 0x8086 pch_did:	 0x284 pch_rid:		 0x0
self.vid:	 0xffff self.did:	 0xffff self.rid:	 0xff
````
With lspci returning:
````
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:9b61] (rev 0c)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:0284]
````
and *nothing* for SPI or LPC.

I guess my patch above is wrong but I'm still confused to why self.vid/self.did doesn't identify the host bridge on the HP box but maybe that's because the platform is really not supported at all. I'll open a different issue because I already added too much noise on this one.

I'll redo a clean test on a supported platform to make sure #907 fixes the initial issue with the XML config installation.

yaperez-anssi, 2020-05-11:
So I can confirm that #907 does fix the installation issue, it works fine on the ThinkPad X280 platform I have.

yaperez-anssi, 2020-07-13:
I think this one can be closed, there were two separate issue (the installation issue fixed with #907, the unsupported platform fixed by #912 )

closed 2020-07-13 by yaperez-anssi