chipsec/chipsec #912: Unsupported platform: HP ProBook 450 G7 (platform 8086:9B61)

yaperez-anssi, 2020-05-11:
Hi, I'm not sure what's the right way to report an unsupported platform and the information I should provide. As indicated in #905 I'm using Chipset as a way to validate configuration for hardware acquired by the French administration (see https://www.ssi.gouv.fr/en/guide/hardware-security-requirements-for-x86-platforms/ and https://github.com/ANSSI-FR/chipsec-check).

A hardware I'm currently testing (HP ProBook 450 G7) is unsupported on master (`2cea2f4`), I've tried running with -i and it gives the attached [log](https://github.com/chipsec/chipsec/files/4608744/450g7.log) and especially:

```
ERROR: Unsupported Platform: VID = 0x8086, DID = 0x9B61, RID = 0x0C
ERROR: Platform is not supported (Unsupported Platform: VID = 0x8086, DID = 0x9B61, RID = 0x0C).
```

I can try to dig up some Intel documentation in order to improve support but are there specific steps I should do?

yaperez-anssi, 2020-05-11:
At first sight it seems that:
````
diff --git a/chipsec/cfg/8086/cml.xml b/chipsec/cfg/8086/cml.xml
index f57ebcd..3736b89 100644
--- a/chipsec/cfg/8086/cml.xml
+++ b/chipsec/cfg/8086/cml.xml
@@ -12,6 +12,7 @@
     <sku did="0x3E34" name="CometLake" code="CML" longname="CometLake v1 U4 Core" detection_value="806EC" />
     <sku did="0x3E33" name="CometLake" code="CML" longname="CometLake v1 U6 Core" detection_value="806EC" />
     <sku did="0x9B51" name="CometLake" code="CML" longname="CometLake v1/v2 U6 Core" />
+    <sku did="0x9B61" name="CometLake" code="CML" longname="CometLake" />
   </info>
 
   <!-- #################################### -->
````
would be enough. I'm unsure what to put in the longname though. The CPU is [i5-10210U](https://ark.intel.com/content/www/us/en/ark/products/196449/intel-core-i7-10510u-processor-8m-cache-up-to-4-90-ghz.html).

BrentHoltsclaw, 2020-05-11:
That change looks correct and should allow chipsec to run.  For the longname CometLake U should be enough.

yaperez-anssi, 2020-07-10:
Since the code was merged and a new release  done, I think this one can be closed. Thanks!

closed 2020-07-10 by yaperez-anssi