chipsec/chipsec #932: 4318 error

82ila82, 2020-06-13:
Hi,
I'm totally new to chipsec.
Tried both "install and run as package" and running chipsec in place without installing it.

the command:
sudo python chipsec_util.py spi info -i 

the result: 
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.5.1
[CHIPSEC] OS      : Linux 4.15.0-101-generic #102-Ubuntu SMP Mon May 11 10:07:26 UTC 2020 x86_64
[CHIPSEC] Python  : 2.7.17 (64-bit)

****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
ERROR: 4318

Can anyone point me in the right direction?
Thanks

BrentHoltsclaw, 2020-06-15:
Did you build the driver prior to the install? https://github.com/chipsec/chipsec/wiki/Installing-CHIPSEC-in-Linux

It is not recommended to run with the -i option for that command (that is most likely the cause of the error if you have built the driver).

82ila82, 2020-06-16:
Yes, I built it (under Mint 64-bit).
The only difference with your link here above was that I installed setuptools using Synaptic (and not with pip install setuptools)

BrentHoltsclaw, 2020-06-16:
do you get any more output when running with the -d flag?

82ila82, 2020-06-20:
Running without the -i switch yields to the same 4318 error.
Just tested with "sudo python chipsec_util.py spi info -d" (without the "" of course) and the result is once again the 4318 error, written in red color.

jujialin, 2020-12-16:
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version : 1.5.7
[CHIPSEC] OS      : Windows 10 10.0.18362 AMD64
[CHIPSEC] Python  : 2.7.18 (64-bit)

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API
ERROR: 4130

i have this error，anybody ever meet？？

frinzell, 2020-12-16:
Is the Error 4318 from a non-Intel platform?

skater-boy, 2020-12-17:
Model name:          AMD Phenom(tm) II X4 945 Processor

frinzell, 2020-12-17:
Thanks, that seems to explain it the error.  Currently, there is no AMD support.

frinzell, 2021-02-22:
Appears answered.  Closing.  Reopen if needed.

closed 2021-02-22 by frinzell