chipsec/chipsec #942: help me to analyze the summary result of ChipSec

Azad-Readhat, 2020-06-29:
Hi dears
I've done the test of ChipSec on a HP Proliant DL360P G8 server in order to find the security level.
I provided the log-file result of the ChipSec in the following.
[result_server.txt](https://github.com/chipsec/chipsec/files/4844009/result_server.txt)

Indeed, I'm confused to analyze **NOT implemented and NOT applicable modules**
Please help me to find answer of these questions
1- What is the reason of these two different modules??
2- Are these modules threatening?
3- is there any way to fix them?
Tnx to all

BrentHoltsclaw, 2020-06-29:
SKIPPED_NOT_IMPLEMENTED - CHIPSEC currently has not implemented support for this test on this platform.

SKIPPED_NOT_APPLICABLE -The issue checked by this module is not applicable to this platform. This result can be ignored.

To check if a module should be implemented, the registers/device being checked must be present on the system and hold the same security implications.  They would need to be assessed on a case by case basis to determine whether or not they are not applicable or should be run on the system.  For the uefi related modules, the system must be booted in UEFI mode for them to run.

Azad-Readhat, 2020-07-01:
dear Brent Holtsclaw
I'm grateful of your helpful description

Azad-Readhat, 2020-07-04:
> SKIPPED_NOT_IMPLEMENTED - CHIPSEC currently has not implemented support for this test on this platform.
>
> SKIPPED_NOT_APPLICABLE -The issue checked by this module is not applicable to this platform. [...]

dear Brent Holtsclaw
Thank you for your help.
How can I test the security of these NOT implemented modules by ChipSec?
Is there any software to do that?

BrentHoltsclaw, 2020-07-07:
For the cpu/pch within the log?

Azad-Readhat, 2020-08-02:
> For the cpu/pch within the log?

Hi BrentHoltsclaw
Yse
for example in a platform, ME (Management Engine) module isn't implemented by ChipSec. in this case (/for other modules that ChipSec cannot to evaluate them) is there any valid method to test?

Sae86, 2022-05-04:
Hi @Azad-Readhat, do you still need assistance with this issue? Are you still getting the same not implemented modules with latest releases?

Sae86, 2022-07-15:
Closing as inactive. Please reopen if needed.

closed 2022-07-15 by Sae86