chipsec/chipsec #953: Blackarch - DKMS make error

PaintedBlck, 2020-07-05:
Dear ChipSec development team,

since some time, chipsec produces an error when trying to install it with DKMS (or when doing a system update). Some help from your side would be welcome.
- Is there a manual way to build and install that can do the trick?
- I join some error information below (if you need more info, please let me know)

Kind regards,
PaintedBlck
 
_**The system**_:
- x64 hardware
- Linux BlackArch - 5.7.7-arch1-1

_**Method of installing**_:
- `git clone https://aur.archlinux.org/chipsec-git.git`
- `cd chipsec-git`
- `makepkg -Acs`
- `sudo pacman -U chipsec-git-1.5.2.r2.gaf7fa73-1-x86_64.pkg.tar.xz`

_**The error message**_:
```
==> dkms install --no-depmod -m chipsec -v 964.af7fa73 -k 5.7.7-arch1-1
Error! Bad return status for module build on kernel: 5.7.7-arch1-1 (x86_64)
Consult /var/lib/dkms/chipsec/964.af7fa73/build/make.log for more information.
==> Warning, `dkms install --no-depmod -m chipsec -v 964.af7fa73 -k 5.7.7-arch1-1' returned 10
==> depmod 5.7.7-arch1-1
```

_**The log file /var/lib/dkms/chipsec/964.af7fa73/build/make.log**_ :
```
DKMS make.log for chipsec-964.af7fa73 for kernel 5.7.7-arch1-1 (x86_64)
Sun 05 Jul 2020 03:09:03 PM CEST
make: Entering directory '/var/lib/dkms/chipsec/964.af7fa73/build'
make -C /lib/modules/`uname -r`/build M=/var/lib/dkms/chipsec/964.af7fa73/build clean
make[1]: Entering directory '/var/lib/dkms/chipsec/964.af7fa73/build'
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Leaving directory '/usr/lib/modules/5.7.7-arch1-1/build'
rm -f amd64/cpu.o
nasm -f elf64 -o amd64/cpu.o amd64/cpu.asm
make[1]: Entering directory '/var/lib/dkms/chipsec/964.af7fa73/build'
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
  CC [M]  /var/lib/dkms/chipsec/964.af7fa73/build/chipsec_km.o
  LD [M]  /var/lib/dkms/chipsec/964.af7fa73/build/chipsec.o
  MODPOST 1 modules
WARNING: modpost: could not find /var/lib/dkms/chipsec/964.af7fa73/build/amd64/.cpu.o.cmd >
ERROR: modpost: "kallsyms_lookup_name"[/var/lib/dkms/chipsec/964.af7fa73/build/chipsec.ko>
make[2]: *** [scripts/Makefile.modpost:99: __modpost] Error 1
make[1]: *** [Makefile:1645: modules] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.7.7-arch1-1/build'
make: *** [Makefile:27: chipsec] Error 2
make: Leaving directory '/var/lib/dkms/chipsec/964.af7fa73/build'
```

BrentHoltsclaw, 2020-07-07:
It appears that there is something with the `kallsyms_lookup_name` function and _kernel 5.7.7_.  After a recent update within another OS  a similar problem occurred.  Need to find some time to look into this farther.

yu17, 2020-07-07:
I've seen a fix for the same problem in another project. I'm not sure if it's gonna help but I'm posting it here anyways.

[nbulischeck/tyton-ae7988b](https://github.com/nbulischeck/tyton/commit/ae7988b572c5c4c01830d2af10e79cdf6ed87f19)

PaintedBlck, 2020-07-07:
I'll make a fix and launch a pull request for this problem.

@yu17 - thank you for your input. This is exactly what is needed.

npmitche, 2021-02-27:
I believe this and #972 have been fixed over the last few releases. Please re-open if that's not the case or submit a new issue if other errors occur.

closed 2021-02-27 by npmitche