chipsec/chipsec #955: A windows driver issue when verify active muti-threads above 48 cores. (Cooper Lake SP 4 sockets, 48 cores)

Eathonhsu, 2020-07-08:
Hi, 
We met many test cases ERROR when active multi-threaded tasks. (_ia32cfg_, _memlock_, _cpu_info_ ... etc)
Our products support 4 sockets **48** cores (12 cores * 4 cpus). 96 threads when hyper-threading is enabling.

Just like the title, test cases will get ERROR when active core above 48 cores.
We found this function can't get correct active cores:
https://github.com/chipsec/chipsec/blob/af7fa737eeba2236b58b7cdbe36c74c5909ca30f/drivers/win7/driver.c#L347

We try to two experiments:
1. Disable hyper-threading (active 48 cores/threads), these task are PASSED. 
2. Hard-coding 96 active cores in the driver, these task are PASSED.

Our OS is **Windows Server 2019 Version 1909**.

Sincerely,
Ethan Hsu

frinzell, 2020-07-08:
Thank you.  This is related to how Windows handles processor groups.  They are limited to 64 threads.  `KeQueryActiveProcessorCountEx(ALL_PROCESSOR_GROUPS)` would be the updated version of this.  This is on the list of items in need of updating.

SeolforHsieh, 2020-07-09:
> Thank you. This is related to how Windows handles processor groups. They are limited to 64 threads. `KeQueryActiveProcessorCountEx(ALL_PROCESSOR_GROUPS)` would be the updated version of this. This is on the list of items in need of updating.

Hi,
It works after replace to `KeQueryActiveProcessorCountEx(ALL_PROCESSOR_GROUPS)`, thank you.

Eathonhsu, 2020-07-09:
> Thank you. This is related to how Windows handles processor groups. They are limited to 64 threads. `KeQueryActiveProcessorCountEx(ALL_PROCESSOR_GROUPS)` would be the updated version of this. This is on the list of items in need of updating.

Hi,
It works after replace to `KeQueryActiveProcessorCountEx(ALL_PROCESSOR_GROUPS)`, thank you.

closed 2020-07-09 by Eathonhsu